steveicarus/iverilog #1048: Assignments to enum with a ternary operator require explicit cast

adogadkin, 2023-12-17:
Hello,

Icarus seems to behave strangely when it encounters enumerated name to be assigned to an enum variable as a part of ternary operator. The following SystemVerilog code

```
module test;
  logic a;
  enum logic { S0, S1 } b;

  always_comb
    b = a ? S0 : S1;
endmodule
```

throws

`test.sv:6: error: This assignment requires an explicit cast.`

message and fails to compile. Casting it explicitly does the trick indeed, but it doesn't seem to make much sense and I couldn't find anything in IEEE1800-2012 that would call for an explicit cast in such assignment.

This seems to happen only when both enum and ?: come together, since the following compiles without any errors:

```
module test;
  logic a;
  enum logic { S0, S1 } b;
  logic c;

  always_comb begin
    if (a) b = S0;
    else   b = S1;
    
    c = a ? '0 : '1;
  end
endmodule
```

I could reproduce this in 12.0 stable and in current master. In all cases -g2012 option was used.

This behaviour was mentioned by jotego in #452, but since the issue itself was unrelated and labeled as Enhancement, I decided to make a new one.

larsclausen, 2023-12-17:
Hi, the issue was also discussed in #1015. Please the the answer there.

adogadkin, 2023-12-17:
Thanks for quick reply!

Somehow I missed that issue during my search.

I faced this in a synthesizable part of my design, where I'm not expecting 'x' to be a possible value. Was your idea of making it into a warning ever considered since then?

larsclausen, 2023-12-17:
No, not yet. A workaround for now is to make your conditional variable `bit` instead of `logic`.

martinwhitaker, 2023-12-17:
A quick survey of other simulators showed that none of them treat this as an error or issue a warning. They happily assign `1'bx` to `b`.

martinwhitaker, 2023-12-17:
However IEEE 1800-2017 does say
> An enum variable or identifier used as part of an expression is automatically cast to the base type of the
enum declaration (either explicitly or using int as the default). A cast shall be required for an expression
that is assigned to an enum variable where the type of the expression is not equivalent to the enumeration
type of the variable.

which would suggest that a cast should be required, even if the conditional variable is 2-state.

larsclausen, 2023-12-17:
The heading of that sections is "enumerated types in numerical expressions". I think that paragraph means "numerical expressions" when it says "part of an expression". Otherwise you couldn't assign a enum constant to a enum variable, since a enum constant is also a expression :slightly_smiling_face:

martinwhitaker, 2023-12-17:
>  Otherwise you couldn't assign a enum constant to a enum variable, since a enum constant is also a expression 🙂

According to https://accellera.mantishub.io/view.php?id=1429, that's just sloppy wording :-)

adogadkin, 2023-12-17:
> They happily assign `1'bx` to `b`.

This makes good sense, the idea of "make an assumption that can not be evaluated -> get indeterminate result" is something I would naturally expect. But if the standard says otherwise, assigning 1b'x quietly is even worse way of handling it.

Maybe changing it from error to warning could be a compromise.

adogadkin, 2023-12-18:
After doing some reading about x-optimism and x-propagation I now somewhat understand the reasoning behind the way ternary operator works in SV. A bit confusing that if...else doesn't x-propagate assignments the same way...

I also realize now that switching from error to warning won't make any difference by itself, since we're still trying to assign a non-enumerated value to our enum.

I found yet another similar issue (#280), so guess I'm just +1 in a row of people getting confused how seemingly same piece of hardware ends up simulated differently :)

Thanks for your help in getting this stuff clarified!

martinwhitaker, 2024-01-08:
In the past, when the standard has appeared to say one thing and all major simulators have done something different (but not obviously wrong), we've gone with the herd, so I wouldn't be opposed to doing the same here.

caryr, 2024-01-08:
For a true two-state condition no cast is always perfectly safe. Should we be on the safe side and make this a compile warning for a four-state condition or should there be a dynamic warning when there is a condition that actually evaluates to X? We need to make sure that two-state division, etc. are classified as a four-state condition since division by zero even for two-state variables is specified to give an undefined (X) result.

I personally think a compile warning that can be turned on/off along with a dynamic warning when the expression is four-state and evaluates to X/Z is probably the best solution. I think the only exception to the undefined condition warning is if the two expressions match exactly.

larsclausen, 2025-01-13:
There is a mantis issue about this now. https://accellera.mantishub.io/view.php?id=8482

davekeeshan, 2025-01-13:
I ran into this problem hard in my last project and ended up having to rewrite a significant amount of code, it only failed in icarus and passed in cadence and verilator, this is going back a bit now, probably even predating this issue. 

I respect that the other tools are probably being lazy and "just doing it" but I do think that this should be documented, even to call out that this is the correct way to do it.

Does anyone know if the casting to the enum type will work in all those tools too? Cadence, synopsys, modelsim?