steveicarus/iverilog #1052: Performance issue for a simple generated design with XOR reductions

flaviens, 2023-12-20:
Hi there!

I noticed that the simulation performance on a simple generated design based on XOR reductions is extremely low in comparison with usual Icarus simulations. Specifically, the speed of `iverilog` is satisfactory, but `vvp` is extremely slow.
The execution of a single evaluation takes more than 2 minutes.

[Here are the sources, with a Readme to reproduce the slow simulation.
](https://github.com/steveicarus/iverilog/files/13728003/slow_xorred.zip)

I'm using the last commit version of Icarus [734f2a0](https://github.com/steveicarus/iverilog/commit/734f2a076bb49bd13ae295cd4e814551338cdbb3).

Thanks!
Flavien

martinwhitaker, 2023-12-22:
Have you looked at the size of the code you are generating? I'm not surprised it is slow.

larsclausen, 2023-12-22:
@flaviens what kind of machine are you running this on? When I run your example it takes 2 seconds.

flaviens, 2023-12-23:
Thank you for your responses!
- @martinwhitaker yes I looked. Simulation is much slower than other random-looking designs of the same size.
- @larsclausen It should be a decent machine (x86, AMD EPYC 7H12).

martinwhitaker, 2023-12-23:
Running your example on my laptop, it takes 2.4 seconds. But if I add an extra step:
````
 in_data = ~in_data;  
#1;  
````
it takes over 2 minutes.

It's a bit difficult to analyse, because your final `assign` statement is a single line of more than 100,000 characters (which my text editor won't even display!) , but I think the problem is you have created a very large concatenation on both the LHS and RHS. Because it is combinatorial logic, each time one element of the input concatenation is updated, the full output concatenation will be re-evaluated and written to the destination.

I would guess the first step is faster because the initial state is `x`, which will short-circuit some of the intermediate evaluations (in a combinatorial network, only changes propagate to the next node).

If you can split up that final assignment, it should run faster.

larsclausen, 2023-12-23:
Part of the problem is the structure of how the signals are connected. There are many signals the have the same input signal multiple times, this leads to a exponential number of update events that are being generated. The following example is a more simplified version of this

```SystemVerilog
module test;

  localparam N = 10;
  localparam M = 8;

  bit x[N];
  bit y;

  assign x[0] = y;

  generate for (genvar i = 0; i < N - 1; i++)
    assign x[i+1] = ^{M{x[i]}};
  endgenerate

  initial begin
    y = 1'b1;
    #1
    y = 1'b0;
  end

endmodule
```

When `x[0]` changes it generates an update event for each of the concat inputs to `x[1]`. When the first input to the concat changes it causes it to be re-evaluated and generate an update event, same for all the other inputs. So you get a total of `M` update events. The same then happens for `x[2]` except that for each of the `M` input events you generate `M` output events, `M*M` total. Eventually you end up with `M**N` events total.

The obvious solution is to not generate multiple events if a input signal that is used multiple times changes. E.g. apply all the concat inputs first and only then generate the output, but I think this would require a fair bit of refactoring.

Another solution can be to have the optimizer recognize these kinds of structures and improve them. Having the same input to reduction operator doesn't change the result (in case of xor they cancel out). So you could have the optimizer remove these duplicated inputs.

martinwhitaker, 2023-12-23:
The reduction operator functors in vvp don't have the usual optimisations (early termination if the input value hasn't changed, calculation and output delayed until the end of the current delta cycle). Adding these optimisations, as in the attached patch, reduced the run time from 130 seconds to 3 seconds on my laptop. Needs more work to check I haven't broken anything in the process - in particular it no longer supports use in an automatic context. For reduction operators it might be more efficient to keep the calculation in the recv_ functions and early-terminate if the result hasn't changed.

[vvp-optimise-reduce.patch.txt](https://github.com/steveicarus/iverilog/files/13760352/vvp-optimise-reduce.patch.txt)

larsclausen, 2023-12-24:
I think that the real problem is with the concat, since it takes multiple inputs. Having the output update delayed for the reduce operation helps with this example, but you could easily construct examples without the reduction operator in there to get the same exponential effect.

I've tried to implement delayed updated for concat itself and it reduces the runtime of the example to effectively 0. See https://github.com/steveicarus/iverilog/compare/master...larsclausen:iverilog:vvp-concat-performance

There are three test cases that fail due to the change in execution order:
 * https://github.com/steveicarus/iverilog/blob/master/ivtest/ivltests/br_gh497a.v
 *  https://github.com/steveicarus/iverilog/blob/master/ivtest/ivltests/pr1002.v
 * https://github.com/steveicarus/iverilog/blob/master/ivtest/ivltests/pr1719055.v
 * https://github.com/steveicarus/iverilog/blob/master/ivtest/ivltests/part_sel_port.v

But these mostly look like they simply have race conditions and can be fixed for the most part by adding a delay before evaluating the value.

martinwhitaker, 2024-01-03:
@larsclausen , agreed, your implementation looks good (once you add your new concat.h file).

PR #1066 (2024-01-20): vvp: Improve concat performance
A concat typically has multiple inputs. Whenever one of the input values
change the output value of the concat is updated and propagated to its
downstream consumers.

When multiple inputs change within the same cycle each input will cause a
update propagation. Depending of the overall structure of the design this
can cause a significant performance penalty.

E.g. the following synthetic structure has a exponential runtime increase
based on the value of N.

```SystemVerilog
reg [N-1:0] x;
generate for (genvar i = 0; i < N - 1; i++)
  assign x[i+1] = ^{x[i],x[i]}; [... cut]
closed 2024-01-20 by larsclausen