steveicarus/iverilog #1053: Undeclared signal - but it works!

logicmonkey, 2023-12-20:
I have a large structural chunk of code that wires up many small blocks in an algorithmic way.
But I forgot to declare all the wires - and yet it still worked.

Here, `t` is not declared.

```vrilog
module mygate (
  input wire a, b,
  output logic y);
  assign y = a ^ ~b; // arbitrary
endmodule

module undeclared;

  logic a, b, c, y; // t is not declared

  mygate u1_mygate( .a(a), .b(b), .y(t));
  mygate u2_mygate( .a(t), .b(c), .y(y));

initial
  for (int ii=0; ii<8; ii++) begin
    {a, b, c} = ii;
    #1;
    $display("%b %b %b %b", a,b,c,y);
  end

endmodule
```

Icarus Verilog version 13.0 (devel) (s20150603-1083-g555a2e70) with -g2012

Shouldn't this throw an error?

larsclausen, 2023-12-21:
This is one of the interesting features of (system)Verilog. There are certain cases where if a identifier has not been declared it will automatically infer it.

iverilog can warn about this by enabling the `-Wimplict` warning (including in `-Wall`).

You can turn it into an error by adding `` `default_nettype none`` at the top of your module.

logicmonkey, 2023-12-21:
Cool - section 6.10 of the 2012 (free) LRM. I should have looked.
I'm not sure some of the strict commercial lint rule decks I've been subject to in the past would be happy, but I guess Verilator will be.

Thanks!

closed 2023-12-21 by logicmonkey

caryr, 2023-12-21:
Yes, I'm sure a linting tool would be configured to report this implicit declaration as an error, but even the commercial simulators should allow it just like Icarus. If you want it to be an error, use the compiler directive @larsclausen mentioned previously. That directive is part of the SystemVerilog language, so using it should make this an error for any simulators.

caryr, 2023-12-21:
Also, I highly recommend using -Wall and cleaning up any warnings that are reported. Icarus follows the standard, but the warnings should help you find some of the issues that linting tools normally complain about.