steveicarus/iverilog #1056: The Iverilog compiler lacks a warning message for mismatched port widths after compilation.

jiangshuirou, 2023-12-29:
Hello, we found a potential Iverilog warning information oversight error.During the analysis of the program, we found that there is a potential issue with the synthesis of “module141” where the width of the port connection “y” does not match the width of the port of “module141”. For instance, the width of the port connection “y” is 22, while the width of the port of “module141” is 18. For such a situation, IVEilog should provide a warning message about the mismatch in port widths.“rtl.v” is our design file.
[12.29.zip](https://github.com/steveicarus/iverilog/files/13790755/12.29.zip)

caryr, 2023-12-29:
What version of iverilog are you using? The current development and V12 both do report warning messages about port connection mismatches (a huge number of them), but the code you provided does not appear to exactly match what you are describing above. It is best for us if you can reduce this down to a simple complete example that exactly matches what you are describing. Icarus is a volunteer project so you investing time to make the debug easier raises the likely hood we can find the issue.

caryr, 2024-01-07:
Have you been able to look into this. Can you provide a simplified/complete example.

jiangshuirou, 2024-01-18:
Hello, the attachment provides a complete description document named “warnning”, which describes the version of iverilog used. For the provided example, please focus on the “module 141”.

[warnning.docx](https://github.com/steveicarus/iverilog/files/13972286/warnning.docx)