steveicarus/iverilog #1073: Runtime error after a xor of concatenations after a latch

flaviens, 2024-01-16:
Hi there!

I stumbled into a runtime error with Icarus, where the output of a latch is wrong after consumed by a xor.
Verilator produces the value that one would expect. To the  best of my knowledge, there is no structural race condition in this circuit.

[Here is a repository to reproduce the issue](https://github.com/flaviens/icarus-b1).
I'm using `Icarus Verilog version 13.0 (devel) (s20221226-331-g77d7f0b8f)`.

Please let me know if you need any more information.

larsclausen, 2024-01-18:
Thanks for all the bug reports.

This is same as #1074. xor with 'x and then the result is 'x and gets converted to 0. Easy to see when changing the type of celloutsig_3z to logic.

Fix will be the same as for #1074. We can probably close this one as a duplicate.

flaviens, 2024-01-18:
Thank you for your response! Then I close this one.

closed 2024-01-18 by flaviens