steveicarus/iverilog #1074: Arithmetics after a flip-flop causes wrong runtime value in specific circumstances

flaviens, 2024-01-16:
Hi there!

Thank you for your contributions with Icarus, it is extremely valuable.
I stumbled into another runtime error with Icarus, where the output of a flip-flop, this time, is wrong after consumed by an arithmetic operator.
Verilator produces the value that one would expect. To the  best of my knowledge, there is no structural race condition in this circuit.

[Here is a repository to reproduce the issue](https://github.com/flaviens/icarus-b2).
I'm using `Icarus Verilog version 13.0 (devel) (s20221226-331-g77d7f0b8f)`.

I would expect this to be related to #1073.
Please let me know if you need any more information.

Thank you!
Flavien

martinwhitaker, 2024-01-16:
The issue can be demonstrated with this much simpler test case:
````
module test();

bit [7:0] a;
bit [7:0] b;
bit [7:0] c;
bit [9:0] d;

assign a[7:4] = 4'd2;
assign b[7:4] = 4'd6;

assign c = b - a;

assign d = { 1'b0, b } - { 1'b0, a };

initial begin
  #0 $display("%b %b %b %b", a, b, c, d);
end

endmodule
````
which outputs
````
01100000 00100000 xxxxxxxx 0000000000

````
This comes about because vvp performs all calculations using 4-state values and relies on values being cast to 2-state at appropriate points when simulating 2-state logic. The casts for the lower (undriven) bits of `a` and `b` are occurring after the functor outputs are used in the subtraction and concatenation operations.

martinwhitaker, 2024-01-27:
Fixed in the master branch.

closed 2024-01-27 by martinwhitaker