steveicarus/iverilog #1082: The sv_literals test is not implemented

steveicarus, 2024-01-21:
This test compiles fine, but generates runtime errors. The test is ivtest/ivltests/sv_literals.v. The error from vvp is:

```
steve@icarus:~/icarus/eda/devel/verilog/ivtest> cat log/sv_literals-vvp-stdout.log 
FAILED -- lsv = 'b1111111111111111 != 'sb1
FAILED -- lsv = 'b1111111111111111 != 'sb11
FAILED -- lsv = 'b111111111111111x != 'sb1x
FAILED -- lsv = 'b111111111111111z != 'sb1z
FAILED -- lsv = 'b1111111111111101 != 'so5
FAILED -- lsv = 'b1111111111101101 != 'so55
FAILED -- lsv = 'b1111111111101xxx != 'so5x
FAILED -- lsv = 'b1111111111101zzz != 'so5z
FAILED -- lsv = 'b1111111111111001 != 'sh9
FAILED -- lsv = 'b1111111110011001 != 'sh99
FAILED -- lsv = 'b111111111001xxxx != 'sh9x
FAILED -- lsv = 'b111111111001zzzz != 'sh9z
```

caryr, 2024-01-23:
This is a bug and likely should be fixed before we release 13.

It looks like the issue is we treat an unsized number as just enough bits to represent the value which is then sign extended during the assignment which is incorrect per the standard. The standard specifies that the value should be zero extended to 32 bits. We could keep the value smaller, but we need to add an extra bit set to 0 for values less than 32 bits that have the top bit set to 1 to prevent the value from being sign extended. We should check other simulators, but the one I checked zero extended up to 32 bits, but for a 32 bit value or larger the high bit was sign extended. For example, when a 36 bit value was assigned to a 64 bit vector.

To me this less than 32 bit versus 32 bit or larger handling gives inconsistent results depending on the value which is why we should check other simulators. Maybe they all do it this way, but it seems like it would be more sane that we add an extra bit no matter the actual width when an unsized signed constant has the MSB set to 1 to prevent sign extending.

larsclausen, 2024-01-23:
I think we should make the larger >= 32 bit behavior here consistent with the `-gstrict-expr-width` handling for unsized decimal literals. E.g. get the same result for `'sd4294967295` and `'shffff_ffff`.

caryr, 2024-01-23:
Sounds reasonable. Consistency is good!

martinwhitaker, 2024-01-25:
For unsized literals used in expressions of width > 32 bits, 1364-2005 table 5-22 states the bit length of an unsized constant number is the same as the bit length of an integer, with the note

>If an unsized constant is part of an expression that is longer than 32 bits and if the most significant bit is unknown (X or x) or three-state (Z or z), the most significant bit is extended up to the size of the expression. Otherwise, signed constants are sign-extended and unsigned constants are zero-extended.

SystemVerilog requires the bit length of an integer to be 32.

Some discussion by the people who draft(ed) the standards:

  https://www.accellera.org/images/eda/sv-bc/11350.html
  https://www.boyd.com/1364_btf/report/full_pr/428.html

Based on that, I say that we should zero-extend up to `integer_width` but no further. If nobody disagrees, I'll fix this.

martinwhitaker, 2024-01-28:
Fixed in the master branch.

closed 2024-01-28 by martinwhitaker