steveicarus/iverilog #1087: Net declarations using built-in types should be limited to 4-state data types

martinwhitaker, 2024-01-27:
Icarus Verilog currently accepts the following net declarations:
````
wire bit [7:0] b;
wire real r;
````
This is illegal according to IEEE 1800-2017 section 6.7.1 which states that a net must be composed entirely of 4-state bits.

I've tested some other simulators and they all rejected these declarations.

If we want to keep these as Icarus extensions, the 2-state case needs some further work to prevent it taking on undefined values when multiply driven.

svenka3, 2024-01-27:
Curious if you want to move that under some futuristic "wreal support" category.

martinwhitaker, 2024-01-28:
`wreal` is already supported if you use `-gverilog-ams` or `-gxtypes`. I wouldn't suggest removing that.

Having found the Icarus extensions documentation (it's in the developer guide, not in the user guide), that states that the compiler should only allow multiple drivers on 4-state nets. Currently the compiler allows multiple drivers on 2-state nets as well, but not on real nets.

caryr, 2024-01-28:
You should be able to tag these nets effectively as uwire to prevent multiple drivers.

It seems like the extensions should be moved into the user documentation.

martinwhitaker, 2024-01-28:
OK, I'll convert non-4-state wires to uwires when extended types are enabled (-gxtypes) and make them an error when extended types are disabled (-gno-xtypes).

martinwhitaker, 2024-01-28:
Fixed in the master branch. I also made it an error to include a data type in a wire or variable declaration unless either the extended data types extension or a SystemVerilog generation are enabled. I've moved the extensions documentation to the usage section and updated and corrected it.