steveicarus/iverilog #1088: Feature request: add output coloring in terminals for keywords like "ERROR"

DeflateAwning, 2024-01-30:
It would be very helpful if FATAL, ERROR, etc. keywords were highlighted so it's faster to identify issues when debugging test benches. These errors are often hidden among `$display(...);` and monitor statements, and are easy to miss.

caryr, 2024-01-30:
Most of the time error messages come from the iverilog stage and other than elaboration task, which Icarus only has preliminary support for, that's all that is really generated in the compiler. For the vvp/runtime stage the opposite should be true where you mostly have user messages and there are seldomly warning/error messages from the program. I'm not against adding something like this because I do agree it would make individual message more visible and easier to notice, but is this:

>  These errors are often hidden among $display(...); and monitor statements, and are easy to miss.

something you actually experience? Do you have example output?

Also, are you asking to only highlight the error/warning messages the program emits or the runtime $fatal()/$error()/$warning() system tasks as well? Both $fatal() and $error() should be the last or close to the last things emitted. I will assume the later. I just want to make sure we know exactly what you are asking for and that it is clearly documented in this request.

DeflateAwning, 2024-02-13:
I'll preface this response with "I'm a noob" and "I'm still learning", so I'm sure there are better ways to do this. 

That said, the project I'm working on is a hash function over at https://github.com/DeflateAwning/eaglesong-verilog/tree/implement1/src (in the implement1 branch).

When I run [run_tests.sh](https://github.com/DeflateAwning/eaglesong-verilog/blob/implement1/src/run_tests.sh), the `$monitor` statements generate a ton of output, but there are assertions which check the values at certain points (roughly every 43 clock cycles). 

Here's an example of the way I have the assertions right now:

```verilog
        if (eval_output_ready !== 1'b1) begin
            $error("Assertion failed: eval_output_ready !== 1'b1 after too long.");
            tb_error_cnt = tb_error_cnt + 1;
        end
```

These are non-fatal `$error` messages that happen in the vvp "exectution" stage, so it'd be nice to be able to see and find them easily with colors.

You say that the `$error` should be "close to the last thing emitted", but if I'm cycling through 100 test cases, and the 10th test fails, but the next 90 succeed, wouldn't it make sense that the failure is pretty far up in the log? I have a counter that keeps track of the failed test count. Overall feels similar to unit testing in software, which I _think_ is what Iverilog is sort of intended for.

caryr, 2024-02-13:
Does adding a wrapper function (e.g. `color_str_red(<message string>`)) to color the message solve the issue? Yes, the whole assertion will not be colored, but having the message in color should give you the grab the eye impact you are looking for. The problem we have adding this to Icarus is it needs to be done in a manner that works with all terminals, but for your tests it just needs to support the terminal you are using.

I do like the idea of adding color support, it's just harder than adding the standard escape sequences since we need to determine the terminal capabilities and then do the best we can to add color or some form of highlighting.

DeflateAwning, 2024-02-13:
I think that'd work yeah! That'd be great! 

I feel like working with color must be at least mostly a solved problem, right?

caryr, 2024-02-14:
I didn't see anything when I did a quick look after you posted originally. Though it was not an extensive search. I agree, I would have expected a somewhat standard library for terminal capability detection and then colorization.

DeflateAwning, 2024-02-14:
Suggestion: We could default to colors, and support the `NO_COLOR` (https://no-color.org) environment variable to disable it? Would that make too many people angry?

Otherwise, looks like ncurses supports it. I think ncurses is a massive library that we probably don't want though, right? [Source](https://stackoverflow.com/questions/39275398/detect-if-cout-outputs-to-a-terminal-which-accepts-colors-properly)

DeflateAwning, 2024-02-14:
Three library options:

1. ColorDebug - Colorful command line output C/C++ macros
2. textus_coloris - C header library for colourised console output
3. sgr - C++ stream manipulators for colored text output and more