steveicarus/iverilog #1095: Question on AMS support

zhengqigao, 2024-02-06:
Hi,

Please excuse me for asking the following question if it sounds repeated. Does the current tool support a Digital-Analog Converter (DAC) or ADC? For instance would it support something like [this Verilog-AMS code snippet for DAC/ADC on Wiki](https://en.wikipedia.org/wiki/Verilog-AMS)?

caryr, 2024-02-20:
Along with the contribution statement that the other report mentions. I believe the transition statement is also not supported. You can model an A/D or D/A in Icarus using real values, but not in the AMS manner.

Kreijstal, 2024-07-12:
> Along with the contribution statement that the other report mentions. I believe the transition statement is also not supported. You can model an A/D or D/A in Icarus using real values, but not in the AMS manner.

how would you connect a custom analog function like transition to iverilog-a