steveicarus/iverilog #1096: Override parameters in top-level Verilog

scorbetta, 2024-02-07:
I would like to override parameters' defaults in my top-level Verilog:

```
module TOP #(
    parameter NUM_INPUTS = 1,
    parameter NUM_OUTPUTS = 1,
    parameter WIDTH = 8,
    parameter FRAC_BITS = 3
)
(
    input wire CLK,
    ...
);
```

using the following command line (I'm using Icarus Verilog 12.0 stable):

```
/usr/local/bin/iverilog -o sim_build/sim.vvp -s TOP -DWIDTH=8 -DFRAC_BITS=5 -DNUM_INPUTS=16 -DNUM_OUTPUTS=8 TOP.v
```

However, I keep falling into an error that makes me think those parameters are *not* updated. As a matter of fact, everything goes as expected if I replace the above defaults with desired values and run `iverilog` w/o any `-D` option.

I've also tried `-P`, but nothing changes.

Any idea?
Thanks

S

martinwhitaker, 2024-02-07:
`-D` defines a macro. Macros are not parameters. When using `-P`, you need to give it the full hierarchical path, e.g. `-PTOP.WIDTH=8`.

Unfortunately the compiler doesn't report an error if you don't provide a valid module path.

scorbetta, 2024-02-07:
@martinwhitaker Terrific! Thanks. Solved

closed 2024-02-07 by scorbetta