steveicarus/iverilog #1097: Array depth defined with the num method of an enum where the variable name is a identical to part of the enum instance name causes an error

abradd, 2024-02-08:
The title is a bit of a mouthful, hopefully an example will make it clear:

``` verilog
module test_mod ();
 
    typedef enum logic [4:0] {ENUM_ELEM1, ENUM_ELEM2} test_enum_t;

    test_enum_t test_mem_addr_e;
    logic test_mem [test_mem_addr_e.num()];
    logic test_mem_int [test_mem_addr_e.num()];

endmodule 
```

Running `iverilog` on the above throws the following error:

```
test2.sv:8: error: No function named `test_mem_addr_e.num' found in this context (test_mod).
test2.sv:8: error: Dimensions must be constant.
test2.sv:8       : This size expression violates the rule: test_mem_addr_e.num()
2 error(s) during elaboration.
```

If you append any text to `test_mem`, as long as does not match `test_mem_addr_e`, no error is generated. 

I tested this using 2948e9a89f706144eab61dffc5642c959a019b8e.

abradd, 2024-02-08:
I tried some other variable name combinations that throw errors:

``` verilog
module test_mod ();
    
    typedef enum logic [4:0] {ENUM_ELEM1, ENUM_ELEM2} test_enum_t;

    test_enum_t test_mem_addr_e;
    logic mem_try [test_mem_addr_e.num()];
    logic test_mem [test_mem_addr_e.num()];
    logic test_ [test_mem_addr_e.num()];
    logic addr_e [test_mem_addr_e.num()];
    logic e_addr [test_mem_addr_e.num()];

endmodule 
```

`mem_try` and `e_addr` aren't exact substrings of the enum instance name.

larsclausen, 2024-02-08:
This is effectively the same as #483. There is currently no support for referencing a variable or wire as part of the dimension specification of another variable or wire. Will keep this open since it provides another example of how this can fail.

abradd, 2024-02-08:
Perhaps I am missing something, but this additional test seems to work for defining dimension specifications of one variable/wire using another variable/wire:

```SystemVerilog
module test_mod ();
    
    typedef enum logic [4:0] {ENUM_ELEM1, ENUM_ELEM2} test_enum_t;

    test_enum_t test_mem_addr_e;
    logic [3:0] value [test_mem_addr_e.num()];

    initial begin
        $display("value: %b, %b, (%0d bits)", value[0], value[1], $bits(value));
    end
    
endmodule 
```

No errors are thrown and `value` has the number of bits I would expect. If I dump to VCD I also see the relevant signals in GTKWave.

martinwhitaker, 2024-02-09:
The problem is that the compiler does not elaborate the variable declarations in the order in which they are declared. The parser stores the variable declarations in a std::map container, keyed by the variable name. So when the elaborator iterates over the stored declarations, it gets them in alphabetical order. So if your array name follows your enum name in alphabetical order, your code will compile successfully. If it doesn't, it won't.

#483 has the additional problem that generated scopes are elaborated after the parent scope.

martinwhitaker, 2024-04-06:
Fixed in the master branch.

closed 2024-04-06 by martinwhitaker