steveicarus/iverilog #1099: unsigned port connection sign extends

zachjs, 2024-02-11:
The following example:
```Verilog
module top;
    reg [3:0] inp1;
    reg signed [3:0] inp2;
    wire [4:0] out1, out2;
    initial begin
        $monitor("%b %b %b %b", inp1, inp2, out1, out2);
        #1 inp1 = 4'b1111;
        #1 inp2 = 4'b1111;
    end
    mod m1({inp1}, out1);
    mod m2({inp2}, out2);
endmodule

module mod(
    input [4:0] inp,
    output [4:0] out
);
    assign out = inp;
endmodule
```
Outputs:
```
iverilog-extend-bug.v:10: warning: Port 1 (inp) of mod expects 5 bits, got 4.
iverilog-extend-bug.v:10:        : Padding 1 high bits of the port.
iverilog-extend-bug.v:11: warning: Port 1 (inp) of mod expects 5 bits, got 4.
iverilog-extend-bug.v:11:        : Padding (signed) 1 high bits of the port.
xxxx xxxx 0xxxx xxxxx
1111 xxxx 01111 xxxxx
1111 1111 01111 11111
```

Per the warning and the sim output, `{inp2}` is incorrectly treated as signed in this context. The same issue occurs for `$unsigned(inp2)`, `unsigned'(inp2)`, and `inp2[3:0]`, but not for, e.g., `{inp2[3:2], inp2[1:0]}`. All of these expressions should be treated as unsigned.

I am running `Icarus Verilog version 13.0 (devel) (s20150603-909-g83db6915)`.

martinwhitaker, 2024-02-12:
The concatenation is being optimised away when the input port expression is synthesised to create the structural netlist. In this case we still need a wrapper to convert the signed value to unsigned.

martinwhitaker, 2024-02-24:
Fixed in the master branch.

closed 2024-02-24 by martinwhitaker

zachjs, 2024-02-25:
This issue still exists for the other examples: `$unsigned(inp2)`, `unsigned'(inp2)`, and `inp2[3:0]`. Can you please take a look?

martinwhitaker, 2024-02-25:
Also fixed.