steveicarus/iverilog #1104: iverilog doesn't exist with an error code if an include file is missing.

zapta, 2024-02-23:
I am using iverilog via apio on Mac OS. Today I run a build, iverilog reported correctly that an include file is missing  but exit with status code of 0.  Is there a way to tell iverilog to treat it as an error and exit with an error code?

```
$ iverilog -V
Icarus Verilog version 12.0 (devel) (s20150603-1135-g6b127432-dirty)

$ iverilog -B "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" -o sensor_timing_tb.out -D VCD_OUTPUT=sensor_timing_tb -D NO_ICE40_DEFAULT_ASSIGNMENTS "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v" deserializer.v main.v queue.v queue_pusher.v sensor_timing.v sensor_timing_tb.v

sensor_timing_tb.v:2: Include file testing.i not found

$ echo $?
0
```

martinwhitaker, 2024-02-24:
This is because the error comes from the pre-processor, which is a separate program whose output is piped into the main compiler. So we can't see the pre-processor exit status. To fix this we need to indicate the error in the pre-processor output stream. I think we could do this using a `pragma directive.

martinwhitaker, 2024-02-24:
OK, I see there's already a mechanism for passing the error count via a comment. It's just not being used for some failure cases.

zapta, 2024-02-24:
Thanks a lot Martin!

martinwhitaker, 2024-02-24:
Fixed in the master branch.

closed 2024-02-24 by martinwhitaker