steveicarus/iverilog #1105: Add support for `assert( l_c === 1'b0 );` syntax

DeflateAwning, 2024-03-01:
It would be awesome if the standard SystemVerilog assert syntax was supported.

```verilog
// currently not supported
  assert( l_c === 1'b0 );

// current work-around
  if( l_c === 1'b0 ) ; else $error;
```

martinwhitaker, 2024-03-01:
Are you using a very old version of iverilog? I added support for immediate assertions in 2019.
````
module test;

logic l_c;

initial begin
  assert(l_c === 1'b0);
end

endmodule
````
````
% iverilog -g2009 test.v
% vvp a.out
ERROR: test.v:6: 
       Time: 0  Scope: test
````

DeflateAwning, 2024-03-01:
I wasn't specifying an appropriate language version. I didn't realize that it wasn't introduced until `-g2009`. Thanks!

closed 2024-03-01 by DeflateAwning

martinwhitaker, 2024-03-02:
It's available with any SystemVerilog generation. But iverilog defaults to standard Verilog.