steveicarus/iverilog #1107: Support implicit named port connections

shareefj, 2024-03-06:
Hi,

I've had a search for "implicit port" and nothing came up but I can't believe I'm the first to report this so perhaps my search foo is poor.

Is there any planned support for implicit port connections of the form:

```
modname instname (
  .port
);
```

rather than having to use the regular:

``` modname instname (
  .port (port)
);
```

larsclausen, 2024-03-06:
This should work when using SystemVerilog mode e.g. `iverilog -g2012`

shareefj, 2024-03-06:
Thanks. I was using that switch but with v12_0 release.  Should I be using latest main branch?

larsclausen, 2024-03-06:
This should be support in v12. What is the error message you are getting?

martinwhitaker, 2024-03-08:
Implicit connections are only allowed for port connections, not for parameter assignments. Closing as invalid.

closed 2024-03-08 by martinwhitaker