steveicarus/iverilog #1110: When we used Iverilog for the synthesis process, we encountered an abnormal error problem.

yelen103, 2024-03-13:
When we used Iverilog for the synthesis process, we encountered an abnormal error problem. Our `Subsystem.sv` file motivates its file `Subsystem_tb.sv` to fail to compile and test through Iverilog. We passed the simulation on both vivado and quartus compilers using the same source files. We found the root cause of the code, there was an error passing a value in the subfile that is now being referenced. If we comment out the value passing of the ff function in cfblk1, the error will no longer occur. We think this may be due to the reset value recovery issue of ff. We hope you can help eliminate our problem and solve it.
[Iverilog_1.zip](https://github.com/steveicarus/iverilog/files/14585017/Iverilog_1.zip)

martinwhitaker, 2024-03-14:
The error screenshot in your `doc.docx` shows your compiler command line is wrong. `-y` takes a directory name, not a file name. And as you are using SystemVerilog, you need to pass a suitable `-g` option.

Using this command
````
iverilog -g 2012 Subsystem_pkg.sv Subsystem.sv Subsystem_tb.sv
````
I see a number of syntax error messages. Looking at the first one, that is because you are using a literal array assignment pattern which contains a replication operator. That SystemVerilog feature is not yet supported in Icarus Verilog.

You say you are using iverilog for the synthesis process. I assume that is just a translation error, as Icarus Verilog is a simulator, not a synthesis tool.

caryr, 2026-11-06:
This was fixed by #1715

closed 2026-11-06 by caryr