steveicarus/iverilog #1111: vpi_put_value() at T0 is overridden

toddstrader, 2024-04-05:
As discussed here:
https://github.com/cocotb/cocotb/pull/3825

See example:
[t0_put.tar.gz](https://github.com/steveicarus/iverilog/files/14889724/t0_put.tar.gz)

Basially:

- I bootstrap via the VPI
- add a callback for 0 time in the future (issue also appers if callback is for `cbStartOfSimulation`)
- assign a signal via `vpi_put_value()`
- confirm that the value was put
- register a callback for some point in the future
- something (????) changes the signal to `z`
- callback is executed and now the signal has `z` instead of the value that was put

I think the `something` is Icarus, but cannot be sure.  Also, I do not observe this issue on other simulators:
https://edaplayground.com/x/Qa7Z

Also also, if my initial callback is set for some non-zero time into the future, I do not observe the rouge `z` being assigned.

Is my initial callback coming in before Icarus is done setting pre-initial values for all the signals?  And if so, should that setup happen before my callback?

martinwhitaker, 2024-04-05:
This looks like #1041. Testing your example with the head of the master branch gives
````
BOOTSTRAP
SYSTEM FUNCTION REGISTERED
DOING VPI
SIGNAL vpiFullName = dut.signal
SIGNAL = 123
THE DUT
signal = 123
==> signal = 123
CALLED BACK
SIGNAL vpiFullName = dut.signal
SIGNAL = 123
````

toddstrader, 2024-04-05:
Ah, bummer.  I tried to search for anything related but did not find that issue.  Yeah, looks like `HEAD` is happy.  Thanks for checking.

closed 2024-04-05 by toddstrader