steveicarus/iverilog #1123: Arrays can't be used in sensitivity lists

mzannoni, 2024-05-10:
I tried to compile with `iverilog` the following code:  
```
localparam InWdt = 8;
localparam CntWdt = 16;
reg [CntWdt-1:0] count [InCnt-1:0];
...<some more code here>...

always@(count) begin

...<and more code here>...
```
and I got the following error:  
```
../sv_src/tb_countones_tst.sv:46: error: Array count needs an array index here.
../sv_src/tb_countones_tst.sv:46: error: Failed to evaluate event expression 'count'.
2 error(s) during elaboration.
```

Further details:
* the whole file can be found [here](https://github.com/mzannoni/dig_it_all/blob/0000619b4e37ecb6e8412c5f245b2e23557bcd46/sv_src/tb_countones_tst.sv) (specifically created to test this)
* the command I used: `iverilog -g2012 -o test ../sv_src/tb_countones_tst.sv` (from inside a working sub-dir)

The compilation can be successfully completed (and the simulation run) if the line is corrected by _manually_ expanding explicitly the sensitivity list into the array components:
```
always@(count[0], count[1], count[2], count[3]) begin
```

It seems that Icarus can't accept an array in the sensitivity list. The expected behaviour is that each signal of the array is part of the sensitivity list. I verified that this can be correctly understood by `verilator`.

Is this a missing feature that is planned to be implemented?

I think this should be fixed (added?) to address some specific cases, for instance:
* if the array is very large (the actual use case where I needed it would require me to manually write down 32 signals in the sensitivity list...) then a lot of manual coding (and related error-prone code duplication) is required
* if the array is defined with a parameter, it is essentially not possible to implement this (or is there a way to write an array expansion in a sensitivity list?)

martinwhitaker, 2024-05-10:
This is not legal Verilog. IEEE 1800-2017 section 9.4.2 states:

> A variable used with the event control can be any one of the integral data types (see 6.11.1) or string. The
variable can be either a simple variable or a ref argument (variable passed by reference); it can be a
member of an array, associative array, or object (class instance) of the aforementioned types.

> Event expressions shall return singular values. Aggregate types can be used in an expression provided the
expression reduces to a singular value. The object members or aggregate elements can be any type as long as
the result of the expression is a singular value.

So no, this is not a missing feature, and there is no plan to implement it.

mzannoni, 2024-05-11:
Thanks for your reply, I understand.
I had no idea this was not "legal" Verilog, the tools I've used on this module previously (like Verilator or the FPGA tool-chain from Xilinx, for instance) never complained and interpreted as expected.

This does not help me simulate such a circuit, though.
Any suggestion on how I could use Icarus Verilog to compile and simulate such a circuit?

I'm thinking maybe using a `generate` statement and properly made module... But this would mean significant longer code, but properly still safe.

mzannoni, 2024-05-11:
> This is not legal Verilog. IEEE 1800-2017 section 9.4.2 states:
>
>
>
> So no, this is not a missing feature, and there is no plan to implement it.

I actually checked in the standard (IEEE 1800-2023, so the one that superseded the one you quoted, no clue if it's different in this specific section though...) and I'm not sure I agree with your conclusion that this is not legal Verilog...
I'd like to understand your conclusion, though.

Specifically, with specific mention to the sections you quoted, why would this:
`always@(count[0], count[1], count[2], count[3]) begin`
be legal, while this other not:
`always@(count)`
?

It would maybe be a matter of interpretation that an array should be expanded to an OR of its components, but I wouldn't see why an OR of its component would not be legal (when it's explicitly typed, it's actually understood by `iverilog`).

Anyways, please note I'm not trying to convince to implement anything, but mostly try to figure out:
* how are other Verilog compilers in reference to Icarus
* how could I implement this in Icarus

And if this back and forth results in a feature proposal than even better :wink:

martinwhitaker, 2024-05-11:
> I actually checked in the standard (IEEE 1800-2023, so the one that superseded the one you quoted, no clue if it's different in this specific section though...) and I'm not sure I agree with your [...]

First read section 6.4 for the definitions of singular and aggregate types. Then reread the text I quoted (the first paragraph has been removed in 1800-2023, but the second paragraph is unchanged - that's OK, because the two paragraphs were saying the same thing in different words). `count` has an aggregate type. `count[0]` has a singular type.

> Anyways, please note I'm not trying to convince to implement anything, but mostly try to figure out:
> 
>     * how other Verilog compilers work, in relation to Icarus (to know the _best_ tool for a given need)

You could try your code with other compilers at https://www.edaplayground.com/

>     * how could I implement this in Icarus (I would like to use Icarus in this specific project of mines)

Use `always @*` or a generate loop.

If you want some background reading, try https://accellera.mantishub.io/view.php?id=1010 (and the links to earlier discussions). You'll find lots of disagreement about what the semantics of `always @*` should be, but agreement that `always @(count)` is illegal.

mzannoni, 2024-05-13:
Thanks @martinwhitaker for pointing me to this tool.
I'm experimenting now with it and indeed, it seems that only `verilator` (not present in the ones available on [edaplayground](https://www.edaplayground.com/)), among the ones I've personally tried, _understands_ the code I mention in this issue.
I briefly tried with Cadence and Siemens compilers (just with the default options) and it seems they also can't compile it.
I'm now digging a bit deeper into this, and also finding an equivalent an legal implementation.

martinwhitaker, 2024-05-18:
Closing as the current behaviour complies with the IEEE standard, and `always @*` provides the desired functionality.

closed 2024-05-18 by martinwhitaker