steveicarus/iverilog #1125: Clock of register is out of sync between RTL vs Yosys genereted netlist

zhergarvi, 2024-05-15:
There is asynchronous reset register in RTL which runs on clock fabric_clk_div from PLL. After synthesizing this RTL, the netlist contains same clock that goes to DFFRE but both clocks are not in sync with each other. The netlist clock get triggered before RTL clock. Below is the screenshot:
[image]
**Note:** After running this design on Verilator, both clocks were in sync.
**Steps to reproduce:**
1. iverilog -g2012 *.v
2. vvp a.out
**Files:** [test.tar.gz](https://github.com/steveicarus/iverilog/files/15318014/test.tar.gz)

martinwhitaker, 2024-05-15:
This is a race in your code. Look at how the PLL LOCK signal is generated and used.

martinwhitaker, 2024-05-18:
Closing as this is not an Icarus Verilog issue.

closed 2024-05-18 by martinwhitaker