steveicarus/iverilog #1128: Request: add `exit` alias to `vvp`'s terminal interface

parker-research, 2024-05-21:
When running `vvp` to execute a file, pressing Ctrl+C puts you into a terminal-like interface with the following commands:

```
> help
Commands can be from the following table of base commands,
or can be invocations of system tasks/functions.

cd       - Synonym for push.
cont     - Resume (continue) the simulation
finish   - Finish the simulation.
help     - Get help.
list     - List items in the current scope.
load     - Load a VPI module, a la vvp -m.
ls       - Shorthand for "list".
pop      - Pop one scope from the scope stack.
push     - Descend into the named scope.
step     - Single-step the scheduler for 1 event.
time     - Print the current simulation time.
trace    - Control statement tracing (on/off) when the code is instrumented.
where    - Show current scope, and scope hierarchy stack.

...
```

Adding an `exit` alias for `finish` would be very helpful so that it aligns with a standard *nix approach to "getting out of tools when at a prompt". 

Additionally, Ctrl+C and Ctrl+D at this prompt should both probably do the `finish` action (that is, kill `vvp`).

caryr, 2024-05-21:
Adding an alias should be relatively simple. Switching the signal handling at the prompt is possible, just needs to be handled correctly so they only have the `finish` functionality at the prompt.

caryr, 2024-06-17:
I've added the alias for exit to development. Tweaking the signals needs more work.

caryr, 2024-12-29:
I doubt I will make ^C an alias for finish since accidentally typing two ^C could cause you to exit instead of going to the command prompt. Say with a slow machine you type the first one and nothing happens so you type it again. I'll think about ^D at the command prompt acting like finish.

parker-research, 2025-01-06:
Ctrl+D would be great, yeah! Very standard across shells