steveicarus/iverilog #1129: Please tag an official `v13.0.1`, or `v13.0.1-rc1`, or something

parker-research, 2024-05-21:
The current `v13` is much better and includes significant improvements above the latest official release (`v12`). I want my research group to be able to use a version of this tool with a well-known tag (so that it can be referenced in communication, papers, etc.) which is later than the `v12` release, but that is not just "whatever's in main that day".

The release doesn't have to be the most stable version around (hence the `v13.0.1-rc1` proposal), but will allow users to discuss and reference a specific version of the software which is on a path towards release.

caryr, 2024-05-21:
Yes, we have been planning a release for some time (see #1058), but most of us work at very demanding jobs during the day so we have not had much time available for Icarus or a new release for a few years. Maybe we should look at using development snapshots that can be done more regularly. We try to set a very high bar for actual releases, but a snapshot can be less stringent in the requirements.

parker-research, 2024-05-21:
Makes sense, and I appreciate all the effort of the development team. I don't mean to sound demanding.

I maintain that snapshot/development preview/other releases would be hugely beneficial. 

No release will be perfect, but the amount of changes-per-release seems way, way higher than the amount of changes between release of most open source software projects. At the very least, I think this project would benefit from minor releases (e.g., v12.1, v12.2, v13.1, v13.2, etc.) in general.

Kreijstal, 2024-07-12:
probably minor releases would be the way to go

martinwhitaker, 2024-07-12:
Minor releases would require someone to spend the time back-porting bug fixes to the stable branch. I used to do that as a matter of course for any bugs I fixed, but the changes to the test suite mean it is no longer a simple process, so I've given that up.

DeflateAwning, 2024-07-13:
You could just not do backports? Require that people go to the latest release for all bug fixes. This is the only way I've seen community-driven open source projects do it.

Kreijstal, 2024-07-13:
shouldn't distro mantainers do backports tho? maybe the devs of icarus are also distro mantainers or something