steveicarus/iverilog #1130: Executing multiple `vvp` instances of different files in parallel on the same computer causes vvp to hang

parker-research, 2024-05-21:
I haven't been able to create a precise reproduction of this, but the gist is that running even a very small module and associated testbench (from the verilog-eval suite) causes `vvp` to hang indefinitely. 

Version: OSSCAD nightly build on 2024-05-20.

caryr, 2024-05-21:
Is this multiple `vvp` processes running on the same machine for the same person or just multiple `vvp` processes? What OS is this on? I think I have previously run multiple `vvp` processes at the same time for the same user so this could be a more recent change.

martinwhitaker, 2024-05-21:
I regularly run multiple `vvp` processes on the same machine at the same time. I have never had a problem doing this.

parker-research, 2024-05-21:
This issue was actually me misattributing #1131. Thanks for the reinforcement!

closed 2024-05-21 by parker-research