steveicarus/iverilog #1133: Adding signal output will cause abnormal simulation output in iverilog

WeneneW, 2024-05-21:
In the following two code snippets, y is the output wire.
```
assign y = {
            forvar19
           };
```
```
assign y = {
            wire7,
            forvar19
           };
```
The original design of the two statements generated `syn_vivado.v` after synthesis in Vivado, and I simulated the synthesized code. 

Theoretically, if all other conditions remain the same, the simulation output in the `forvar19` section should be consistent, meaning that the lower bits of `y` should be the same for the `forvar19` value in both cases.

However, when I simulated using iverilog, with all other conditions being equal, the two produced different simulation outputs.
 (both `wire7` and `forvar19` I constructed are 2-bit).

[image]
The left side only includes the `forvar19` output as shown (highlighted in yellow for the `forvar19` output section); 
the right side includes the output with the `wire7` signal as shown (highlighted in yellow for the `forvar19` output section). 

Inconsistencies appeared on lines 6, 7, and 14. 

Therefore, I used Vivado's simulation tool to simulate the synthesized output of the two statements, and the lower bits of the `forvar19` output were consistent.

[image]
The left side only includes the `forvar19` output as shown (highlighted in yellow for the `forvar19` output section); 
the right side includes the output with the `wire7` signal as shown (highlighted in yellow for the `forvar19` output section). 

**Comparing the outputs of the two simulation tools mentioned above, I think that the issue might be caused by adding signal output, leading to abnormal simulation results in iverilog.** 

My original designs are as follows:
- Output including only `forvar19`:

```
module top  (y, clk, wire3, wire2, wire1, wire0);
  output wire [3:0] y;
  input wire [(1'h0):(1'h0)] clk;
  input wire signed [(3'h4):(1'h0)] wire3;
  input wire [(3'h6):(1'h0)] wire2;
  input wire signed [(2'h2):(1'h0)] wire1;
  input wire signed [(2'h2):(1'h0)] wire0;

  wire signed [1:0] wire7;
  wire signed [(3'h4):(1'h0)] wire5;
  wire [(2'h2):(1'h0)] wire4;
  reg [(3'h4):(1'h0)] reg13 = (1'h0);

  reg signed [1:0] forvar19 = (1'h0);
  reg signed [(2'h2):(1'h0)] forvar15 = (1'h0);

  assign y = {   
               forvar19
                };
  assign wire4 = wire0;
  assign wire5 = ((((8'ha6) ? wire2 : wire0) ?
                     ((8'ha2) ? wire1 : wire1) : (8'ha5)) - wire4);

  assign wire7 = $signed((wire1[(2'h2):(2'h2)] ?
                     $unsigned(wire3) : $unsigned(wire5)));

  always
    @(posedge clk) begin
      if (wire7[(1'h1):(1'h1)])
        begin
          if ((-((forvar15 ? reg13 : wire5) ?
              wire4[(1'h0):(1'h0)] : ((8'ha5) ? (8'h9e) : forvar15))))
            begin
[... 518 lines cut ...]
│   │   ├── cells_xilinx_7.v
│   │   ├── README.md
│   │   └── cells_yosys.v
│   ├── rtl.v
│   ├── syn_vivado.v
│   ├── vivado.log
│   ├── vivado_main
│   ├── vivado_testbench.v
│   └── vvp_vivado.log
```
Descriptions:

- **1/ or 2/**:  These two documents correspond to the two different designs mentioned above

- **data/**: Contains module files referenced by Vivado synthesis.
  - **cells_cmos.v**
  - **cells_cyclone_v.v**
  - **cells_verific.v**
  - **cells_xilinx_7.v**
  - **cells_yosys.v**
- **rtl.v**: Original design.
- **syn_vivado.v**: File synthesized using Vivado.
- **vivado_testbench.v**: Testbench file.
- **vvp_vivado.log**: Simulation results using Icarus Verilog (iverilog).
- **vivado.log**: Simulation results using Vivado.

martinwhitaker, 2024-05-21:
This is almost certainly a race in your design.
Have you noticed that Vivado has synthesised a latch for `y` and no flip-flops?

WeneneW, 2024-05-22:
> This is almost certainly a race in your design. Have you noticed that Vivado has synthesised a latch for `y` and no flip-flops?

Thank you very much for your replyamidst your busy schedule. 
To eliminate the latch, I modified the code: 
I changed the two instances of `for (forvar19 = (1'h0); (forvar19 < (1'h0)); forvar19 = (forvar19 + (1'h1)))` to `forvar19 <= 1'h1;`. 
The latch disappeared, and the output of both versions was consistent when using iverilog. 

However, for the example above where iVerilog simulation results don't match up, when I used `Aldec Riviera Pro 2023.04` on `edaplayground`for simulation, the results were consistent with Vivado simulation. It seems that these tools handle such latch conditions more effectively.

output only includes the `forvar19` [https://www.edaplayground.com/x/GxX4](https://)

Output `wire7` and `forvar19` [https://www.edaplayground.com/x/Q2iW](https://)

I hope that iVerilog can address this issue in the future and become a better simulation tool. Thank you for your contributions to the FPGA field. Best wishes to you.

martinwhitaker, 2024-05-22:
Icarus Verilog handles latches perfectly well. But if there is a race condition in your design, you can't expect all simulators to give the same result. If you haven't done so already, read the "Scheduling semantics" section in the LRM (IEEE 1364 or IEEE 1800), particularly the "Nondeterminism" and "Race conditions" sub-sections.

Closing as this is not an Icarus Verilog issue.

closed 2024-05-22 by martinwhitaker