steveicarus/iverilog #1135: Changing Fixed VVP Path for Prebuilt Icarus Verilog on a Restricted System

Nado15, 2024-06-12:
Hi!

I downloaded a prebuilt version of Icarus Verilog because I am working on a restricted machine where I don't have sudo rights. Specifically, I can't write to directories like /bin, /opt, etc.

I got the prebuilt version from [this GitHub repository](https://github.com/stnolting/icarus-verilog-prebuilt).

I was able to compile some Verilog code using this prebuilt version, but I am unable to run the compiled code. I added all executables to `$PATH`, but I encounter the following error when I try to run the code:

`bash: ./top: /opt/icarus_verilog/bin/vvp: bad interpreter: No such file or directory`

As I mentioned, I can't copy vvp to /opt nor can I create symlinks.

I tried the following solution suggested by ChatGPT:

`export VVP=/path/to/vvp`

However, this did not resolve the issue. Is there another solution to this problem?

caryr, 2024-06-12:
You should be able to compile from source and use a local directory for the bin, lib, etc. Of course you need to have the appropriate development tools available.

Also from what you have shown, have you tried running the compiler output using `vvp <output file from the compiler>`. You should also be able to update the compiler output to point to the actual `vvp` executable.

martinwhitaker, 2024-06-30:
No response, so assuming @caryr's answer solved the problem and closing.

closed 2024-06-30 by martinwhitaker

Nado15, 2024-06-30:
Sorry, I thought I have already answered! Yes it helped, thanks a lot!