steveicarus/iverilog #1138: Non-interactive mode of vvp (-n, -N) seems to be broken

msinger, 2024-06-15:
Since commit 95810b2 vvp always enters interactive mode when the simulation gets interrupted by CTRL-C (SIGINT), even if it was given the -n or -N option on command line.

I usually call vvp with -N, so I can quickly stop it, make some changes to my Verilog code, and start it again. Now I have to enter $finish again and again, which is annoying.

Was this behavioral change intended? I don't understand which change in this commit causes that.

martinwhitaker, 2024-06-15:
I always use `-n`, and that is still working for me. But `-N` is indeed broken.

msinger, 2024-06-15:
Oh, yes, you're right. I was absolutely convinced that I had tested `-n` and it behaved the same as `-N`. I might have messed something up. So `-n` is working correctly for me too, it immediately exits when pressing CTRL-C. Only `-N` is misbehaving then.

martinwhitaker, 2024-06-15:
Fixed in the master branch.

closed 2024-06-23 by larsclausen