steveicarus/iverilog #1139: syntax error on systemverilog

Kreijstal, 2024-06-16:
So, I have the following testbench:
```sv
module student_rlight_tb;
  logic clk;
  logic rst_n;
  logic [7:0] led;
  tlul_pkg::tl_d2h_t tl_d2h;
  tlul_pkg::tl_h2d_t tl_h2d;

  // 50 MHz
  always begin
    clk = '1;

```

I am running it like this:
```
$ iverilog -g2012 -Isrc/rtl/tlul/pkg -o student_rlight_tb.vvp src/fv/student_rlight_tb.sv
src/fv/student_rlight_tb.sv:5: syntax error
src/fv/student_rlight_tb.sv:5: error: Invalid module instantiation
src/fv/student_rlight_tb.sv:6: error: Invalid module instantiation
```
But already at line 5, iverilog says that I have a syntax error...
However this seems to work on the other sims, what is the problem?

martinwhitaker, 2024-06-16:
You haven't compiled the package definition. `-I` just sets the search path for include files; it won't automatically include anything.

We could do with a better error message in this situation, so I'll label this as an enhancement.

Kreijstal, 2024-06-16:
Okay, yeah, I am new to iverilog I think that was the problem I thought `syntax error` meant there was something wrong with syntax itself but it seems it is a semantic error, not a syntactic one, right?

larsclausen, 2024-06-16:
Sorry, unpacked array parameters are not yet supported. See #846.