steveicarus/iverilog #1141: function being classified as non-constant

Adivinedude, 2024-06-27:
when simulating this code
```verilog
function automatic [7:0] f_NaryRecursionGetUnitWidthForLatency;
    input [7:0] base, latency;
    f_NaryRecursionGetUnitWidthForLatency=iterator_NaryRecursionGetUnitWidthForLatency(base,latency,2);
endfunction
function automatic [7:0] iterator_NaryRecursionGetUnitWidthForLatency;
    input [7:0] base, latency, lut_width;
    iterator_NaryRecursionGetUnitWidthForLatency=
        (f_NaryRecursionGetDepth(base,lut_width)<=latency)
            ?lut_width
            :iterator_NaryRecursionGetUnitWidthForLatency(base,latency,lut_width+1);
endfunction
```

I get the following error message
```
    ././toolbox/recursion_iterators.v:222: error: A function invoked by a constant function must be a constant function local to the current module.
    ././toolbox/recursion_iterators.v:322: error: A function invoked by a constant function must be a constant function local to the current module.
    ././toolbox/recursion_iterators.v:309: error: A function invoked by a constant function must be a constant function local to the current module.
```

but when rewritten like this, the error goes away and everything works
```verilog
function automatic [7:0] iterator_NaryRecursionGetUnitWidth;
    input [7:0] base, lut_width, unit, results;
        `define next_level_unit_count (base / lut_width * lut_width == base ? base / lut_width : base / lut_width + 1)
    iterator_NaryRecursionGetUnitWidth =
        (base == 1 )
            ? 0
            : (results + `next_level_unit_count) <= unit
                ? iterator_NaryRecursionGetUnitWidth(`next_level_unit_count,lut_width,unit,results + `next_level_unit_count)
                : (unit - results ) == `next_level_unit_count-1
                    ? base % lut_width == 0 ? lut_width : base % lut_width
                    : lut_width;
endfunction

```

These two code segments perform the exact same function, just written differently. I tried many revisions to get it to build.

martinwhitaker, 2024-06-27:
Please provide a complete test case that demonstrates the issue.

Adivinedude, 2024-06-28:
its rather long and github wont let me upload a .v
this was pulled from my repo history and demonstrates the problem successfully.
I included a define statement to highlight the section of code in question and the solution I found
It could just be a type-o I cant see.
```verilog
`default_nettype none
`define use_failing_code
module example
    #(
        parameter WIDTH     = 4,
        parameter LATENCY   = 4
    )
    (
        input   wire                clk,
        input   wire                ce,
        input   wire    [WIDTH-1:0] I1,
        output                      gate_and
    );

    function automatic [7:0] f_NaryRecursionGetVectorSize;
        input [7:0] base, lut_width;         
        f_NaryRecursionGetVectorSize=iterator_NaryRecursionVectorSize(base, lut_width, 0);
    endfunction
    function automatic [7:0] iterator_NaryRecursionVectorSize;
        input [7:0] base, lut_width, rt;   
        iterator_NaryRecursionVectorSize=
            base==1
                ?rt
                :iterator_NaryRecursionVectorSize(
                    base / lut_width * lut_width == base
                        ? base / lut_width
                        : base / lut_width + 1
                    ,lut_width
                    ,rt + ((base / lut_width * lut_width == base)
                        ? base / lut_width
                        : (base / lut_width) + 1));
    endfunction

    function automatic [7:0] f_NaryRecursionGetUnitWidth;
        input [7:0] base, lut_width, unit;
        f_NaryRecursionGetUnitWidth=iterator_NaryRecursionGetUnitWidth(base, lut_width, unit, 0);
    endfunction
`ifdef use_failing_code
    function automatic [7:0] iterator_NaryRecursionGetUnitWidth;
        input [7:0] base, lut_width, unit, results;
        integer next_level_unit_count;
        begin : block_NaryRecursionGetUnitWidth
            `define next_level_unit_count (base / lut_width * lut_width == base ? base / lut_width : base / lut_width + 1)
            if( base == 1 )
                iterator_NaryRecursionGetUnitWidth = 0;    // overflow condition, requested unit not in range, width = 0 is a valid answer;
            else begin
                if( (results + `next_level_unit_count) <= unit) begin
                    // requested unit is on a different iteration, proceed to the next iteration
                    iterator_NaryRecursionGetUnitWidth = iterator_NaryRecursionGetUnitWidth(
                            `next_level_unit_count,
                            lut_width,
                            unit,
                            results + `next_level_unit_count);
                end else begin
                    // requested unit is on this iteration.
                    if( (unit - results ) == `next_level_unit_count-1 )
                        iterator_NaryRecursionGetUnitWidth = base % lut_width == 0 ? lut_width : base % lut_width;
                    else
                        iterator_NaryRecursionGetUnitWidth = lut_width;
                end
            end 
            `undef next_level_unit_count     
        end
    endfunction
`else
[... 132 lines cut ...]
    always @( posedge clk ) begin
        if( counter == 0 )
            I1 <= I1 + 1'b1;
    end

    initial begin
        $dumpfile("UUT.vcd");
        $dumpvars(0, bug );
        #100 $finish;
    end
endmodule
```
terminal output
```
iverilog -i bug_report_tb.v
bug_report_tb.v:37: error: A function invoked by a constant function must be a constant function local to the current module.
bug_report_tb.v:179: error: `f_NaryRecursionGetUnitWidth' is not a constant function.
bug_report_tb.v:179: error: Cannot evaluate genvar init expression: f_NaryRecursionGetUnitWidth(WIDTH, CMP_LUT_WIDTH, unit_index)
bug_report_tb.v:179: error: `f_NaryRecursionGetUnitWidth' is not a constant function.
bug_report_tb.v:179: error: Cannot evaluate genvar init expression: f_NaryRecursionGetUnitWidth(WIDTH, CMP_LUT_WIDTH, unit_index)
bug_report_tb.v:179: error: `f_NaryRecursionGetUnitWidth' is not a constant function.
bug_report_tb.v:179: error: Cannot evaluate genvar init expression: f_NaryRecursionGetUnitWidth(WIDTH, CMP_LUT_WIDTH, unit_index)
7 error(s) during elaboration.
. . .
```

martinwhitaker, 2024-06-30:
Fixed in the master branch.

closed 2024-06-30 by martinwhitaker