steveicarus/iverilog #1143: incomplete for statement produces no output.

Adivinedude, 2024-07-08:
when I try to skip the 'step assignment' statement in a for loop,  Iverilog reports no errors and vpp does not generate any output. In most other languages, this statement is optional, but I can not find any statement in IEEE 1364-2005 regarding if this is an optional statement or not. This code works properly when building with gowin designer, I have not tested other tool chains.  This was difficult to track down.
```verilog
//`define NOBUG

module bug_module(clk, out);
    input   wire clk;
    output  wire out;
    function automatic integer f_bug;
        input integer value;
        begin
            `ifdef NOBUG
                for( f_bug = 0; value > 0; f_bug = f_bug + 1 ) begin    // No bug
            `else
                for( f_bug = 0; value > 0; ) begin    // bug
            `endif
                if( value == 1 ) begin
                    f_bug = 0;
                    value = 0;
                end
            end
        end
    endfunction
    localparam lp = f_bug(1);
    reg r = 0;
    always @(posedge clk) r <= ~r;
    assign out = r;
endmodule

module test;
    reg     clk = 0;
    wire    out;

    bug_module UUT( .clk(clk), .out(out) );

    always #1 clk <= ~clk;
    always @(posedge clk) begin
        $display("%b", out);
    end

    initial begin
        $dumpfile("UUT.vcd");
        $dumpvars(0, test);
        $display("starting mux_tb.v");
        #8 $display( "***WARNING*** Forcing simulation to end");
        $finish;
    end
endmodule
```

caryr, 2024-07-09:
Check out SystemVerilog (1800-2023). From the code it looks like the call you have for `f_bug` should always return a constant 0 value (calculated at compile time). I've not run this using Icarus so I don't know if it is working correctly.

caryr, 2024-07-09:
FYI making the terms optional was added in 1800-2012.

Adivinedude, 2024-07-09:
Is there a simple way to produce a warning of a runtime error? Instead of just a graceful termination.
I have encountered this !
`no output` problem before, this is just the first time I was able to isolate the cause

martinwhitaker, 2024-07-09:
When I try to compile your example, the compiler segfaults during the elaboration phase and generates no code. So there is nothing for vvp to run.

Adivinedude, 2024-07-09:
When using vs code with the oss-cad-suite i do not see a seg fault. I will try running iverilog directly  from the cmd prompt(Windows 11, version 23H2 64bit)

Adivinedude, 2024-07-09:
So I tried powershell and command prompt. How did you see the segment fault?

martinwhitaker, 2024-07-10:
> So I tried powershell and command prompt. How did you see the segment fault?

On my Linux system the segfault is reported in the system log. But I can also tell the compile has failed because:
1. `iverilog` returns a non-zero exit code
2. `iverilog -v` exits after printing `ELABORATING DESIGN`.

Note that I have pushed a fix for the segfault to the master branch.

closed 2024-07-11 by Adivinedude

martinwhitaker, 2024-07-13:
I have added checks in the compiler to report an error if any of the initialisation statement, termination condition, or step statement are null when a generation earlier than 1800-2012 is selected.