steveicarus/iverilog #1144: $display not working inside function during non clocked execution

Adivinedude, 2024-07-13:
after building newest commit today, Thank you BTW. its really helped me out alot... sry. after building and copying it over to the oss-cad-suite, I noticed that functions no longer $display when called outside of an always block or initial block.

```verilog
// Icarus Verilog Parser/Elaborator version 13.0 (devel) (s20221226-529-gcb6544fac)
module bug_module(clk, out);
    input   wire clk;
    output  wire out;

    function automatic integer f_bug;
        input integer value;
        begin
            $display("function: %1d", value);
            f_bug = value + 1;
        end
    endfunction

    localparam lp = f_bug(0);   //first call of function. does not print
    reg r = 0;
    integer temp;
    initial temp = f_bug(1);    // prints
    always @(posedge clk) r <= ~r;
    always @(posedge clk) temp <= f_bug(temp); // repeated calls of function print.
    assign out = r;
endmodule

module test;
    reg     clk = 0;
    wire    out;

    bug_module UUT( .clk(clk), .out(out) );

    always #1 clk <= ~clk;

    initial begin
        $dumpfile("UUT.vcd");
        $dumpvars(0, test);
        $display("starting bug_report_tb.v");
        #8 $display( "***WARNING*** Forcing simulation to end");
        $finish;
    end
endmodule
```
```
>iverilog bug_report_0_tb.v
>vvp a.out
function: 1
VCD info: dumpfile UUT.vcd opened for output.
starting bug_report_tb.v
function: 2
function: 3
function: 4
function: 5
***WARNING*** Forcing simulation to end
bug_report_0_tb.v:36: $finish called at 8 (1s)
```

larsclausen, 2024-07-13:
This is expected. When you initialize a localparam this is done at compile time hence there is no print at runtime

Adivinedude, 2024-07-13:
That makes since, it was $displaying in version 13.0 (devel) (s20221226-524-gdc6f9f204-dirty), and I just upgraded to get the constant function bug patch and noticed it was not acting the same.

closed 2024-07-13 by Adivinedude