steveicarus/iverilog #1148: Simulation results discrepancy

LoSyTe, 2024-07-21:
During our synthesis process with Yosys, we generated two synthesis files using default and custom synthesis parameters. Theoretically, these two synthesis files should have identical logical functionality, and thus their functional simulation results should be consistent. However, when we used Icarus Verilog for simulation, we observed discrepancies in the final simulation results. We believe that this inconsistency is not due to the Yosys synthesis files but is instead caused by Icarus Verilog's handling of these files. We seek the community's assistance in addressing this issue.

my design file（rtl.v） as well as the optimization sequence as follows：

[Uploading project.zip…]()

The test script is：

    yosys -p "
        read_verilog rtl.v         
        synth
        write_verilog syn_yosys.v"
    iverilog -o wave_1 -y syn_yosys.v yosys_testbench.v
    vvp -n wave_1 -lxt2 >> file1.txt
    sed -i 's/wave_1/wave_2/g' file1.txt;
    mv syn_yosys.v old_syn_yosys.v
    
    yosys -p "
        read_verilog rtl.v         
    hierarchy; proc; opt_dff -nodffe; opt_share; opt_clean; abc;
        write_verilog syn_yosys.v"
    iverilog -o wave_2 -y syn_yosys.v yosys_testbench.v
    vvp -n wave_2 -lxt2
However, as shown in the red box in the figure below, the simulation results are inconsistent. We look forward to your next reply.

Using Yosys' default synthesis process，the first line of output is:
[image]

Using Custom optimization sequence，the first line of output is:
[image]

larsclausen, 2024-07-23:
In the example that you posted `wire4` is not driven, which is then assigned to `y` and hence you see the `X`. So I'd say this is expected. Not sure if maybe with the different optimizations yosys drives the signals differently.

caryr, 2024-08-26:
This seems like a synthesis bug. Have you verified the designs are actually 100% matching? Have you looked into the comment Lars made? Also the `syn_yosys.v` file is missing so we cannot actually test this as expected.

caryr, 2024-09-09:
Closing since there has been no response in few weeks. Please reopen if you have pertinent details.

closed 2024-09-09 by caryr