steveicarus/iverilog #1151: During simulation, iverilog incorrectly assigned a value to a variable that was not driven.

zjb-ljm, 2024-07-27:
When using ` iverilog `, ` vivado `, and ` quarters ` to simulate the same composite file ` Subsystem. v `, theoretically, the same simulation results are expected on the three simulation tools. However, the simulation results of ` iverilog ` are different from the other two tools. We will add the source code in the attachment. The following are our simulation results in three simulation tools. It can be clearly observed that the 'hdl_out' variable has consistent simulation results in 'vivado' and 'quarters', but' iverilog 'shows inconsistent simulation results.
[image]
` iverilog `simulation results.
[image]
` vivado `simulation results.
[image]
 ` quarters `simulation results.

`It seems that iverilog has incorrectly assigned a value to a variable that is not driven. It is evident in vivado and quarters that hdl_out assigns x at 100ns, but iverilog provides a fixed value.

Our source files are as follows, where Subsystem. v is our main file.
[source.zip](https://github.com/user-attachments/files/16397807/source.zip)

caryr, 2024-08-07:
The source file you uploaded is not a zip file.

zjb-ljm, 2024-08-07:
I have re uploaded it, could you please confirm it again.

zjb-ljm, 2024-08-07:
Sorry, is there a problem with my understanding? I have uploaded the Verilog design file and testbench file in. zip format as per your request. Do you need anything else?
Look forward to your reply.
[image]

martinwhitaker, 2024-08-07:
Given the strong similarity to #1156, including some almost identical source files, my suspicion is that this is a training course exercise, so I'm reluctant to solve it for you. I suggest you first work on reducing the source code to a minimal test case that demonstrates the issue, then check for race conditions in the code that could lead to different results on different simulators.

If you can provide a simple example that demonstrates that Icarus Verilog is not behaving according to the LRM (IEEE 1364 or IEEE 1800), we will of course look at that.

caryr, 2024-08-07:
Yes, the original was a RAR masquerading as a zip. I wanted things consistent. Thank you for updating the file labeled as a zip to actually be a zip file.

I agree Martin, except I was possibly going to debug it enough to confirm Icarus is working as expected. My guess, like you is this is some kind of training. Possibly to demonstrate that you can legally get different results with different simulators if your SystemVerilog code does not take non-determinism into account. If that's not the case this certainly should be something added for an intermediate to advanced SystemVerilog course and something you'll never get from vendor sponsored training.

Part of learning SystemVerilog is struggling with the problems. Learning how to reduce a design while debugging to isolate the actual issue is a very good skill to develop. Most of my bug submissions for the commercial simulators I use every day are less than 10 lines of code and also reference specific sections of the standard that support why I believe the simulator is not functioning correctly. Yes, sometimes we have to dump the whole design on an application engineer to debug, but this is for a paid tools that has the budget to support specialists like this. We are just part time developers and most of the time we don't even have time to work on the simulator any more, so you spending some extra time to develop a skill releases us to focus on bugs when we do have a few spare minutes/hours to devote to Icarus tasks.

caryr, 2024-08-10:
I looked at this quickly and the code is inconsistent with your comments.

Hdl_out is always driven from a register that is reset properly at time zero and produces the expected results. Like Martin said earlier we need a reduced test file that clearly demonstrates where iverilog is producing incorrect results. If there is something specific that you expect to load an undefined value into the register we need that information.

caryr, 2024-08-26:
See #1161 where I have confirmed this is not an issue in Icarus using that code. Since this is not a complete example I am closing this ticket.

closed 2024-08-26 by caryr

caryr, 2024-08-26:
This is also being closed since the user has deleted their account and they are no longer engaged in the conversation.