steveicarus/iverilog #1153: Save and Restore simulation state

AruneshKma, 2024-08-02:
Hi,
I want to save the simulatiion state. then later restore it and resume the simulation, similar to the one found in verilator ( https://veripool.org/guide/latest/simulating.html?highlight=save#save-restore ), but I couldn't find a feature that'll support this in iverilog. Could you suggest a way in which this can be done in iverilog.

Thanks,
Arunesh

caryr, 2024-08-02:
Icarus doe not support this functionality. Why do you need to stop/save simulation and then restart it?

AruneshKma, 2024-08-05:
Thanks for getting back, it's to run the reset sequence once, save state and use this state to avoid rerunning in every test. Will restore the saved state in the beginning of every test.