steveicarus/iverilog #1156: During simulation, iverilog mistakenly assigned a value to an undriven variable.

llb-del, 2024-08-06:
We are practicing the basic function of' verilog', so we use three different tools,' iverilog',' vivado' and' quartus', to simulate the same comprehensive file' Subsystem.v' to verify the simulation results of our model. Theoretically, the same verilog project should have the same simulation results in three different simulation tools, but we found that the results were not the same after verification. Among them, the simulation results of' vivado' and' quartus' are consistent, while' iverilog' is inconsistent with the simulation results of the other two tools.
[image]
`quarters`simulation results.
[image]
`iverilog`simulation results.
[image]
`vivado`simulation results.

Specifically, as shown in the following figure, the difference lies in' 0ns-100ns'. It can be clearly seen in the tools of' vivado' and' quartus' that the assignment of' hdl_out' is' x', while in' 0ns-50ns', a fixed value is given in the tool of' iverilog', and after '50ns', it is. Iverilog' seems to have incorrectly assigned a variable that is not driven. Currently for the same.

Our source files are as follows, in which Subsystem.v is our main file.
[source.zip](https://github.com/user-attachments/files/16510248/source.zip)

martinwhitaker, 2024-08-06:
This is suspiciously similar to #1151. Is this a training course exercise?

You should try to reduce the code to the simplest test case that still reproduces the issue, and then look for race conditions in the code.

caryr, 2024-08-07:
The source file you uploaded is not a zip file.

caryr, 2024-08-26:
See #1161 where I have confirmed this is not an issue in Icarus using that code. Since this is not a complete example I am closing this ticket.

closed 2024-08-26 by caryr