steveicarus/iverilog #1161: During simulation, hdl_out was incorrectly assigned a value.

zhu-jinbo, 2024-08-25:
We are practicing the basic functions of Verilog. To verify the simulation results of our model, we are using three different tools, namely iverilog, vivado, and quartus, to simulate the same synthesis file, Subsystem. v, in order to validate our model. In this Verilog code, a module called Subsystem is defined, which implements a series of numerical calculations and logical operations. Internally, the module processes input signals (clock, reset, enable) by instantiating multiple submodules and some computational logic. The data is stored and updated through two register arrays (cfblk185_deg and cfblk184_deg), and cfblk153_out1 is calculated as the difference between cfblk184_out1 and cfblk177_out1. Then, the module cfblk157 generates cfblk157_out1, which is assigned to Hdl_out to generate an 8-bit output signal Hdl_out.
          In theory, the same Verilog project should have the same simulation results in three different simulation tools, but after verification, we found that the results were not the same. Among them, the simulation results of 'vivado' and 'quartus' are consistent, while the simulation results of' iverilog 'are inconsistent with the other two tools. As shown in the following figure, the difference is that after '0ns', it can be clearly seen in the' vivado 'and' quartus' tools that the assignment of 'hdl_out' is' x ', but in the' iverilog 'tool, a fixed value is given` It seems that iverilog has incorrectly assigned a value to a variable that is not driven. At present, we believe that the inconsistency in running the same Verilog project on different tools is a potential defect of the iverilog tool.
Running process of iverilog: We simulate the Verilog code by entering the commands ` iverilog - o wave - y Subsystem. v Subsystem. tb. v ` and ` vvp - n wave - lxt2 ` (source file: ` Subsystem. v `, excitation file: ` Subsystem. tb. v `), and finally enter ` gtkwave wave. vcd ` to view the waveform.
Iverilog version number: Icarus Verilog version 12.0 (develop) (s20150603-1110-g18392A46).
                                        GTKWave Analyzer v3.3.108 (w)1999-2020 BSI.

[image]
`iverilog` simulation results Waveform.
[image]
`quartus` simulation results Waveform.
[image]
`vavido`simulation results Waveform.

Our source files are as follows, where Subsystem. v is our main file.
[source_1.zip](https://github.com/user-attachments/files/16739554/source_1.zip)

caryr, 2024-08-26:
This looks to be exactly the same as #1156 and #1151. iverilog is working 100% as defined by the standard and the other tools are not matching the functionality specified in the standard (though this could be how they have been configured). I verified this using an industry standard tool (configured to follow the standard) and manually looking at the code. I will not give an answer to what you are missing since that is likely part of the assignment.

You may want to focus on fixing the floating signals in your design and once everything is driven correctly I expect all the simulators will give the same result.

caryr, 2024-08-26:
I'll also add, even if these tools are configured in an alternate manner as I suspect, they should not produce an undefined value on Hdl_out all the time. There are cases where it should be 1 so they likely have an actual bug in their alternate configuration. This expectation was also verified with the industry standard tool.

caryr, 2024-09-09:
Closing since there has been no response in few weeks. Please reopen if you have pertinent details.

closed 2024-09-09 by caryr