steveicarus/iverilog #1167: inout ports driver collisions handled improperly

tarik-ibrahimovic, 2024-09-03:
This issue is reproducible by following the instructions listed in [this](https://github.com/chili-chips-ba/openCologne/tree/main/2.Simple--1--PSRAM) repo, and origins from an another [issue](https://github.com/chili-chips-ba/openCologne/issues/25#issuecomment-2308363721) listed there.

The issue is that 2 drivers try to access the same signal - `inout wire io_psram_dataX`, while there are no conflicts, iverilog ignores one of the drivers (`psram_sim`) and leaves the signal `z` when the first driver (`psram`) is assigning it to be.

[image]

chili-chips-ba, 2024-09-03:
@steveicarus any example you can offer about simulating `inout` ports and multi-drop, tri-state buses with `iverilog`?

Tarik's example above simulates fine with commercial tools on _EDA Playground_ and, surprisingly, even with nominally 2-state Verilator.

martinwhitaker, 2024-09-08:
From the information supplied, I can't seewhat the issue is. In the source code view, you point to a line in 2.sim/models/psram_sim.sv that generates the `oe` signal for that module, based on `psram_state`. But in the waveform view `psram_state` is `20`, which is neither of the states that causes `oe` to be asserted.

Your second yellow arrow on the waveform view points to `o_state` and `out_enable`, but they belong to the 1.hw/psram.sv model.

tarik-ibrahimovic, 2024-09-10:
Thank you for taking a look, I had a bug in my sim model actually, it's fine now.

closed 2024-09-10 by tarik-ibrahimovic