steveicarus/iverilog #1169: Warning of missing drive by constant 0

jiangshuirou, 2024-09-06:
I am compiling an RTL design file using Iverilog, and after the compilation, Iverilog has issued several warning messages. However, these warning messages are not comprehensive.Analyzing the program, we have identified that there is a potential issue with the top-level design port y[27], where the port is driven by a constant 0. In regards to this situation, Iverilog should issue a warning message about design top has port y[27] driven by constant 0.
[Relevant-documents.zip](https://github.com/user-attachments/files/16905805/Relevant-documents.zip)

martinwhitaker, 2024-09-07:
That is legal Verilog. Why should a simulator issue a warnng about it?

jiangshuirou, 2024-09-09:
During the Vivado synthesis process, the warning “design top has port y[27] driven by constant 0” typically indicates that the port y[27] in the design is being driven by a constant value of 0 without being connected to any other signals or logic. Possible reasons include:
（1）Unused port: The port is not connected to any other signals or logic in the current design, so the synthesizer defaults it to constant 0.
（2）Default value: There might be a default initial or reset value set to 0 in the Verilog code, which is not overridden by any other logic.
（3）Incorrect connection: There could be an error where the port was supposed to be connected to dynamic logic but is mistakenly connected to a constant 0.
（4）Ambiguous design intent: The designer may have intentionally set the port to 0 but did not clearly comment on this in the code, leading to a warning from the synthesizer.
Although this is valid Verilog syntax, a warning is issued to alert the programmer to pay attention to the aforementioned issues.

caryr, 2024-09-09:
You did not answer Martin's question:

> That is legal Verilog. Why should a **simulator** issue a warning about it?

Your reply confirms this is a synthesis related warning:

> During the Vivado **synthesis** process, the warning “design top has port y[27] driven by constant 0” typically ...
> Although _**this is valid Verilog syntax**_, a warning is issued to alert the programmer to pay attention to the aforementioned issues.

A simulator is not expected to issue the same warnings as a synthesis tools.

caryr, 2024-09-16:
Will you be responding to our questions? If not we will close this after giving you some more time to respond.

caryr, 2024-09-26:
No further input provided so closing this issue.

closed 2024-09-26 by caryr