steveicarus/iverilog #1178: Support repeat concatenation nets

mkorbel1, 2024-10-25:
I ran into this error while testing some new [ROHD](https://github.com/intel/rohd) SystemVerilog generation for `wire`s.  The goal is a module which can support a bidirectional replication operation for nets.

> sorry: I do not know how to elaborate repeat concatenation nets.

It looks like it's coming from here:
https://github.com/steveicarus/iverilog/blob/d8c3c51ab1190ed3fb26540d7de6177f83e0e75b/elab_net.cc#L54-L58

Here's a small testbench that reproduces the issue and should pass if the feature were implemented.

```SystemVerilog
module net_connect #(parameter WIDTH=1) (w, w); 
inout wire[WIDTH-1:0] w;
endmodule

////////////////////

module ReplicateMod (
inout wire [3:0] bus,
inout wire [7:0] replicated
);
net_connect #(.WIDTH(8)) net_connect (replicated, ({2{bus}}));
endmodule : ReplicateMod

module tb;
logic [3:0] bus;
wire [7:0] replicated;
wire [3:0] wire__bus; assign wire__bus = bus;
ReplicateMod dut(.bus(wire__bus), .replicated(replicated));
initial begin
#1
bus = 4'h3;
#2
if(replicated !== 8'h33) $error($sformatf("Expected replicated=00110011, but found replicated=0x%x with inputs {bus: 0011}", replicated));
#8
bus = 4'hc;
#2
if(replicated !== 8'hcc) $error($sformatf("Expected replicated=11001100, but found replicated=0x%x with inputs {bus: 1100}", replicated));
#8
$finish;
end
endmodule
```

Are there plans to support this in Icarus Verilog?  I didn't see any other open issues tracking this, sorry if it is a duplicate.

martinwhitaker, 2024-11-03:
There are no development plans for Icarus Verilog - it is mostly in maintenance mode, with any enhancements coming as and when the individual maintainers have the time and enthusiasm.

Repeat concatenations are already supported, but not on the LHS of a continuous assignment. That is explicitly illegal for continuous assignment statements, and more vaguely implied for module output and inout port connections.  I'm not sure why there is this restriction, as if you replace '{2{bus}}` with `{bus, bus}`, that is legal (and supported by Icarus Verilog), but other simulators I have tried refuse to compile your example too.

caryr, 2024-11-09:
I also checked this on my primary work simulator and it clearly reports that output and inout ports cannot have repeat concatenations.

We should check for this and give an appropriate error message.

caryr, 2024-11-10:
I have pushed a patch that reports using a repeat concatenation as a net l-value as an error. There are two ways  your code generator can produce valid SystemVerilog code. Keep track of what is an input and an output or unroll the concatenation loop like Martin suggested previously.

I expect `interconnect` when it is implemented will also have this issue since it also is required to use net l-values.

closed 2024-11-10 by caryr