steveicarus/iverilog #1182: In the initial block, variable declaration results in a syntax error

Noah-S-E, 2024-11-11:
Considering the design below, I have minimized it. I declare variables within the initial block, but it gives a syntax error.

The output is:

```
iverilog -g2005-sv t.v 
t.v:4: syntax error
I give up.
```

```verilog
module fifo_manager();
    initial begin   
      logic [7:0] data;
    end
endmodule
```

martinwhitaker, 2024-11-11:
Correct, that is invalid syntax. You may only declare variables in a named block.

closed 2024-11-11 by martinwhitaker

ajeethakv, 2024-11-11:
With -g2012 it works in Icarus.

martinwhitaker, 2024-11-11:
Sorry, I misread and thought this was with -g2005, meaning standard Verilog.

Using iverilog built from the master branch, it works with -g2005-sv too.

Noah-S-E, 2024-11-11:
Thank you very much for clarifying my doubts.

caryr, 2024-12-29:
I have added the code that reports each declaration as an error to the development branch.