steveicarus/iverilog #1183: Assertion lwid == ivl_signal_width(lsig)' failed

Noah-S-E, 2024-11-12:
I have minimized the design and encountered an assertion error when using a task to assign values to array elements.

The error is as follows (I have upgraded to the latest version of Icarus Verilog `
Icarus Verilog version 13.0 (devel) (s20221226-565-g6c8ed62a5)`):

```
iverilog -g2005-sv t.v
ivl: stmt_assign.c:480: store_vec4_to_lval: Assertion `lwid == ivl_signal_width(lsig)' failed.
```

<br/>

```verilog
module array();
  logic clk;
  logic [1:0] a[2];  

  task task1(output logic [1:0] a);
    begin
      a[0] = 00;  
      a[1] = 01;
    end
  endtask
  
  always @(posedge clk) begin
    task1(a);  
    $display("buff = %h", a); 
  end

endmodule
```

caryr, 2024-11-12:
This is caused by errors in your code that the compiler is not catching.

The error is that the function is declared to take a two bit packed vector, yet you are passing a two element array of two packed bits. It is this mismatch that is causing the assertion. Unfortunately tasks do not currently support unpacked arguments so you would need to pass them individually.

The other error that is actually in the runtime is %h is specified to only print packed elements, but you are passing an array.

It's also worth noting that the task assignments are assigning signed decimal values and not binary which I am assuming was your intention.

I will leave this open since we should report the task call type does not match the task declaration type.

caryr, 2024-11-12:
We should also report that passing an unpacked array to %h, etc. is invalid and not just currently unsupported.

> The integer format specifiers, %h, %x, %d, %o, %b, %c, %u, and %z (uppercase or lowercase), may be used with any of the SystemVerilog integral data types, including enumerated types and packed aggregate data types. These format specifiers can also be used with user-defined types that have been defined (using typedef) to be represented using one of these basic types. **They shall not be used with any unpacked aggregate type.**

Noah-S-E, 2024-11-12:
Thank you very much for clarifying my doubts.