steveicarus/iverilog #1189: support building with meson

EasyIP2023, 2024-12-15:
@steveicarus want to support building ICARUS Verilog with meson/ninja.

Can get the ground going as I have time.

Want to make sure this would be accepted.

closed 2024-12-15 by EasyIP2023

caryr, 2024-12-16:
Since you have closed this then maybe there is no reason for a conversation, but I'm curious why you think we should update the build system? Are there specific issue you have with the existing system based on `make` or are there distinct advantages you see in switching?

EasyIP2023, 2024-12-16:
@caryr No existing method builds without issue.

Personal opinion is that meson is for the modern developer.
The syntax is a lot easier for people to grasp.

But again I closed. Probably, no need for conversation.

caryr, 2024-12-16:
I see no reports of build issues. Do you have specific examples?

I personally build on the last few LTS versions of Ubuntu/Debian, latest GhostBSD/FreeBSD and the UCRT and MINGW64 variants of MSYS2. I currently do not personally have access to MacOS hardware or anything ARM based, though we do use MacOS as part of our CI flow.

The issues I know about is the current VPI MacOS fails in the CI flow after 13 was deprecated and we needed to upgrade to 14 appear to be related to non-functional issues, though I plan to verify and fix this ASAP. I have not tried a compile under ARM so that is an unknown amount of work and likely requires better access to hardware. I would also like to add CLANG64 from MSYS2 which appears to be a linker configuration issue. Also after upgrading to the latest GhostBSD there are some failing tests regarding illegal instructions that need to be tracked down and fixed. This could be a compiler bug or possibly a compiler configuration issue.

I'm not certain if `meson` would resolve any of these.

EasyIP2023, 2024-12-16:
@caryr I'm building in an Ubuntu 24.04 docker container. Builds without issue.

Folks working on meson have made strides to make it cross-platform. 

meson works great on both mac & linux. As to weather it resolves your issue without fiddling with environment is TBD.

On mac believe all I required was

```bash
sudo xcode-select -s /directory/path/to/Xcode.app/Contents/Developer
export SDKROOT=$(xcodebuild -sdk -version 2>/dev/null | grep '^[Path].*\platform/Developer/SDKs/MacOSX.*\.sdk' | cut -d ' ' -f2)
export CC=$(xcodebuild -find clang 2>/dev/null)
export CXX=$(xcodebuild -find clang++ 2>/dev/null)
export CPP="${CXX}"
```

On windows in Git For Windows SDK I required a little fiddling when I utilized meson.

https://anadoxin.org/blog/bringing-visual-studio-compiler-into-msys2-environment.html/

Use envdiff to generate necessary Visual Studio environment variables

```bash
[[ "${OSTYPE}" == "msys" ]] && {
source envdiff-create-file.sh
export MSYSTEM="MINGW64" ; source /etc/profile ; 
export LDFLAGS="-LIBPATH:$(echo "${BUILD_OUTPUT_DIR}/lib" | sed -e 's#/c/#C:\\#g' -e 's:/:\\:g')"
}
```

Meson itself has a ``backend`` flag

```bash
meson setup --backend="vs"
```

Allows you build cross-platform with a few meson.build files giving
you consistent behavior.

Although biased how meson team compares themselves.
https://mesonbuild.com/Comparisons.html