steveicarus/iverilog #1191: Add support for __VA_ARGS__ in a `define MACRO

zaun, 2024-12-26:
I'd like to create some more flexible macros, mostly for logging. Something like:

```
`define LOG(CATEGORY, MESSAGE, ...) \
    $display("[%s] %0t: " MESSAGE, CATEGORY, $time, __VA_ARGS__)
```

but at the moment this isn't supported an I get 

> error: malformed `define directive ('.').

This is just an enhancement request, not a bug.

larsclausen, 2024-12-26:
This would be a nice feature, but since it is not part of the (System)Verilog language itself its probably not going to be implemented.

caryr, 2024-12-27:
Is the following enough?
```
module top;

  `define va_mac(msg, va_msg) $display("[%s] %t: %s", msg, $time, $sformatf va_msg);

  integer v1,v2;

  always @(v1, v2) `va_mac("Log message", ("The value of 'v1' is: %0d and 'v2' is: %0d", v1, v2) )

  initial begin
    `va_mac("Log message", ("--- Start the Log ---") )
    `va_mac("Log message", ("") )
    #1;
    v1 = 0;
    #1;
    v2 = 1;
    #1;
    v1 = 1;
    v2 = 2;
  end
endmodule
```
The trick is making the variadic part a single argument of at least one element that is then handled by `$sformatf`. There are other possibilities using the same technique.

zaun, 2024-12-28:
I didn't know this was possible, my google skills failed me. I'll give this a try. Thank you.

caryr, 2024-12-28:
I assume this will work so I'm going to close this. Reopen if you find something actually broken in Icarus.

closed 2024-12-28 by caryr