steveicarus/iverilog #1193: No option to run vvp in quite mode

pweaverm, 2024-12-30:
Looking at the options for vvp, I see that there is a -l for the log file, but there isn't a corresponding -quiet that allows the simulation to run without pushing the test case text results to the terminal. This can slow the simulation down for larger sims, is it possible to have a -quiet where only the logfile is updated and not the terminal?

martinwhitaker, 2024-12-31:
This seems a useful enhancement and shouldn't be too hard to implement. I'll take it on.

martinwhitaker, 2024-12-31:
This is now implemented in the master branch with the new "-q" flag.

closed 2024-12-31 by martinwhitaker

pweaverm, 2024-12-31:
Hi Martin. I was able to pull the update and verify that the output is quiet with the -q option. Thanks for the quick addition!

One question, what is the background for having the order of the command line options? Why would iverilog or vvp need to have certain options before the code or after? Just curious for my own learning.

Thanks!

martinwhitaker, 2025-01-01:
> One question, what is the background for having the order of the command line options? Why would iverilog or vvp need to have certain options before the code or after? Just curious for my own learning.

This is better explained in the vvp man page than in the on-line documentation. In short, the options before the file name are interpreted by and affect vvp itself, the options after it are passed on to the executed design and VPI modules.