steveicarus/iverilog #1194: No -f option for vvp?

pweaverm, 2024-12-30:
I see there is a limited version of the -f command file for iverilog, but nothing like this for vvp. Is the a possibility of adding the -f option for vvp?

Thanks!

martinwhitaker, 2024-12-30:
Can you explain what advantage(s) this would have compared to using a shell script that invokes vvp with the required arguments?

pweaverm, 2024-12-30:
Hi Martin,
  Typically with XRUN we use the command file to setup the tool with specific tool related/project related options... libraries, tool options etc... then the script that we have only passes parameter or options that are test case specific ie name of test case, path of run directory etc... This decouples the run script from project setup so users don't have to modify the run script with every project. It helps keep the same interface for user no matter what server farm or project they are on.

With XRUN I can pass almost any option to the simulator and since it is a single step compilation, elaboration and simulation tool, I can make one command file with all the necessary options.

I noticed that the command file format only takes library and library path options, so we can't pass the -gxxxx option or anything else for tool configuration. 

Does this help?

Mark

Side note... I'm trying to figure out how to use the extended options for vvp but vvp just says they aren't valid options. Is there something else I'm missing?

Thanks!

martinwhitaker, 2024-12-30:
Hi Mark,

I achieve much the same effect by sourcing a project-wide script from within the run script. A simple example would be:
````
[project.sh]
project_iverilog_options="-g2005 -Wall"
project_vvp_options="-n"
````
and
````
[run.sh]
#!/bin/sh
. project.sh
iverilog $project_iverilog_options $*
vvp $project_vvp_options a.out
````
vvp extended arguments must come after the filename, e.g.
````
vvp -n a.out -vcd
````

pweaverm, 2024-12-30:
Figured out the extended options. Need to place them after the compiled file is listed. This isn't described on the web page documentation, but is in the vvp man output. Just letting you know.  Thanks

pweaverm, 2024-12-30:
Hi Martin,
  Sure creating another project file works, but in our XRUN flow we try to use the command file since it can have all the tool setup options. Of course XRUN has lots of possible options so adding to the command line might not work. limit on the command line length?

Anyway... it was an idea that I thought might help. Thanks for the responses and example of your flow usage. Happy almost new year!

caryr, 2024-12-30:
It seems like a reasonable enhancement, though not a high priority since there are other ways to work around this.

martinwhitaker, 2024-12-31:
WRT vvp extended arguments, the first paragraph of https://steveicarus.github.io/iverilog/usage/vvp_flags.html seemed fairly clear to me, but I've added an additional note at the beginning of the Extended Arguments section. Whilst at it I noticed that not all the vvp flags were documented on that page, so I've copied the missing descriptions from the man page.

pweaverm, 2024-12-31:
Thanks Martin! I see the new comment and the original wording. Missed this original wording because I was focused on the extend args section. The NOTE is a great addition to highlight usage, appreciate it. Thanks!