steveicarus/iverilog #1196: Handling of upper bits for array indices wider than 32 bits

larsclausen, 2025-01-01:
The current behavior for array indices wider than 32 bits is that the upper bits are silently truncated and the lower bits are used as the index E.g.
```SystemVerilog
reg [32:0] i = {1'b1,32'h1};
integer a[2];
a[1] = 10;
$display(a[i]); // -> 10
```

In my opinion this should be treated as out-of-bounds instead. And it is if the index is constant. E.g.
```SystemVerilog
integer a[2];
a[1] = 10;
$display(a[{1'b1,32'h1}]); // -> x
```

But we have a [test](https://github.com/steveicarus/iverilog/blob/master/ivtest/ivltests/signed_a.v#L20-L26) that explicitly checks that upper bits for dynamic indices are discarded.

@caryr its been a while, but you [created](https://github.com/steveicarus/ivtest/commit/5928142d1e9f40f827712b05eb338735d502bd05) that test. Any thoughts on this?

martinwhitaker, 2025-01-02:
I would be inclined to agree with you, but a quick test on some other simulators indicates they also truncate, either silently or with a warning. I found [this discussion](https://www.accellera.org/images/eda/sv-bc/9294.html) which relates, but doesn't reach any conclusion.

caryr, 2025-01-03:
I'll need to take a look. I don't remember the details without more context, but I think we are implicitly assuming that all this eventually gets flattened, normalized and limited to something that is addressed by 32-bits. Maybe the limiting is happening too early. There's also the issue that we do not use types with explicit widths (e.g. uint32_t, int64_t, etc.) so there is some variation depending on the environment.