steveicarus/iverilog #1198: Passing a string as a top-level parameter

davekeeshan, 2025-01-03:
Hi There

I am using `icarus` extensively for a lot of verification work, mainly SoC based.

I compile `icarus` from source and am currently pointing to the head of the design, but I have been seeing this for about a year.

I want to be able to compile some C-code, generate the elf and hex file and pass this hex file into the verilog to use as the contents of its ROM for simulation.  The testbench/design is not changing but the testcase and the location of the ROM file is.  

When I try and pass in the path of the file (`PRELOAD_FILE` here) to the `icarus`:

    iverilog -o sim_build/sim.vvp -s jazard_c7 -g2012 -f sim_build/cmds.f -I/home/gomez/projects/daxzio-jazard_c7/rtl/jazard_c7 -s cocotb_iverilog_dump -Pjazard_c7.XLEN=32 -Pjazard_c7.PRELOAD_FILE=./riscv_build/hellow.hex

It complains:

    <command line>: error: invalid value specified for defparam: jazard_c7.PRELOAD_FILE

I have used relative paths, I have used full paths, none work, and I think it is nothing to do with paths and to do with passing strings.  

I have done this:

    -Pjazard_c7.PRELOAD_FILE=./riscv_build/hellow.hex

and this:

    -Pjazard_c7.PRELOAD_FILE="./riscv_build/hellow.hex"

I know passing in a parameter works as I am also passing in an integer, `XLEN`, and that is getting in and being used correctly (I have changed it as compile time to confirm it is picking up the change).

Here is how the parameters are declared at the top level, 

    , parameter integer XLEN = 16
    , parameter PRELOAD_FILE = ""

I have tried to cast the `PRELOAD_FILE` to a string, but that doesn't work either, exact same complaint (vivado doesn't like that either)

Passing the path explictly at the top of the file itself does work as expected

caryr, 2025-01-03:
You are close. To pass a string the value must include the surrounding `"` and everything you have tried is either omitting the `"` or the shell is consuming the `"`. Try the following:
`-Pjazard_c7.PRELOAD_FILE='"./riscv_build/hellow.hex"'`

caryr, 2025-01-03:
I have update the man page to also include a string override example.

davekeeshan, 2025-01-03:
Yes this was my issue, I have been around linux enough I should have probably tried that, but it is in the bucket of unintutative (that's my excuse).  I have adjusted my flow and am now able to pass in the string successfully.  Thanks for that, closing!

closed 2025-01-03 by davekeeshan