steveicarus/iverilog #1205: vvp exits with "syntax error"

dieterss, 2025-01-21:
When simulating a piece of verilog which has a register array as the shift amount in an expression the compiler produces its output file, but vvp reports "syntax error":
$ iverilog test.v 
$ ./a.out 
./a.out:16: syntax error

This occurs with iverilog version 11.0 (as delivered in debian stable).  Unfortunately I am unable to test latest version, but thought the issue may be of interest none the less.

Test case is as follows:
--cut--
module test;

wire [8 - 1:0] d0;
reg [3 - 1:0] tkq[0:0];
wire [8 - 1:0] val;

assign val = d0 >> tkq[0];

initial
begin
	#1;
	$finish;
end

endmodule
--cut--

caryr, 2025-01-22:
This runs correctly using the latest development. We will need to check older version later.

caryr, 2025-07-01:
This also runs using the stable branch (V12). Closing this since both development and the stable branch work as expected.

closed 2025-07-01 by caryr