steveicarus/iverilog #1206: Feature Request: Flag for redefinition of modules

dpetrisko, 2025-01-22:
Hello! 

Context:
I am looking to use iverilog for GLS of skywater130. I'm using volare to manage the PDK which concatenates all functional simulation models to a single verilog file 'sky130_fd_sc_hd.v'. So then I pass sky130_fd_sc_hd.v along with my netlist to iverilog. This works fine and the library modules are resolved.

However, I would like to use my own definition for a specfic std cell (to work around a simulation modeling issue). So I tried passing the new definition as an additional file to iverilog. I get the following error:
```
iverilog sky130_fd_sc_hd.v bsg_nonsynth_nand2_delay.v testbench.v

bsg_nonsynth_nand2_delay.v:3: error: 'sky130_fd_sc_hd__nand2_1' has already been declared in this scope.
sky130_fd_sc_hd.v:60159:      : It was declared here as a module.
bsg_nonsynth_nand2_delay.v:15: error: Module sky130_fd_sc_hd__nand2_1 was already declared here:sky130_fd_sc_hd.v:60159
```

So it seems like iverilog will not allow me to redefine a module from a previous file. The real proper solution for this would be verilog-2001 configuration mapping, but obviously that's a substantial task. As a stopgap, I would like to request a flag which would cause this redefinition to be a warning and have iverilog take the latest definition as the canonical one. If this feature already exists is some capacity or there is another workaround I could be using, please let me know

Thanks for your time and effort on this awesome project!

martinwhitaker, 2025-01-25:
I thought the `-l` option would do what you want, but on testing I find it doesn't work as I expected.

martinwhitaker, 2025-01-25:
For now the only workaround is to split the library file up into individual module files in a subdirectory and use the `-y` option.