steveicarus/iverilog #1208: Update vpi_user.h

ktbarrett, 2025-01-29:
I noticed that the vpi_user.h file has some out of date definitions, for example `vpi_control` returns `void` in the copy of this file from this repo, while IEEE 1364-2001 and newer show the function returning a `PLI_INT32`. Where does this version of the file come from? Is there some reason to not use the vpi_user.h and the sv_vpi_user.h definition from the IEEE standards in the repo?

martinwhitaker, 2025-02-09:
I don't know where the vpi_user.h file came from. It was added to the Icarus Verilog source code in March 2001. It includes Icarus-specific functionality and implementation details, so we can't simply replace it with the definition contained in the IEEE standards.

ktbarrett, 2025-02-09:
@martinwhitaker If that's the case then can we change the `vpi_control` function to return an `int` to match the spec? It should be backwards compatible with user source code that previously assumed no result.

Strangely, this returns the opposite of what most other VPI functions do.

[image]

martinwhitaker, 2025-02-09:
Yes, I think this is a bug and should be fixed.

svenka3, 2025-02-13:
> Strangely, this returns the opposite of what most other VPI functions do.

Interesting observation. Which VPI functions return 0 on success? While I didn't read through LRM, a quick chatGPT conversation gave me the following:

> vpi_chk_error(vpiHandle errHandle)

> Returns 0 if there are no errors (success).
> Returns 1 if an error is present (failure).
> This is opposite of most other VPI functions.
 
Commonly used ones such as vpi_get returns 0 on failure. So maybe vpi_control was done in similar way to remain consistent within VPI world? 

P.S. not a VPI expert

martinwhitaker, 2025-07-05:
Whilst looking at fixing this, I see that vvp currently supports two identical functions, vpi_control() and vpi_sim_control(). From the comments in vpi_user.h, vpi_sim_control was the name originally proposed by the VPI Task Force, but it was changed to vpi_control by the time 1364-2001 was published. Given that was around 25 years ago, I propose we now remove vpi_sim_control(). @caryr, any objection to this?

caryr, 2025-07-06:
`vpi_sim_control()` is still defined for Verilog-AMS (2023) so it should be kept. Though it is defined to return a `bool` instead of a `PLI_INT32`.

caryr, 2025-07-08:
As long as you only return 1 or 0 no one will likely notice and if they do we can hope the new VAMS version that is supposed to be based on SystemVerilog will be available by then and has resolved this discrepancy.

martinwhitaker, 2025-07-08:
Fixed in the master branch. However, testing exposed another bug. `vpi_control(vpiFinish,...)` and `vpi_control(vpiStop,...)` should respond to the diagnostic level passed as the second parameter (in the same way as `$finish` and `$stop`), but they ignore it.

closed 2025-07-08 by martinwhitaker