steveicarus/iverilog #1209: VPI on Windows

nipscernlab, 2025-02-10:
I've been trying to run vpi in windows by following the
instructions discussed in
http://iverilog.wikia.com/wiki/Using_VPI.

I get this error when compiling: iverilog-vpi hello.c

Compiling hello.c...
hello.c: In function ‘hello_compiletf’:
hello.c:3:33: warning: unused parameter ‘user_data’ [-Wunused-parameter]
    3 | static int hello_compiletf(char*user_data)
      |                            ~~~~~^~~~~~~~~
hello.c: In function ‘hello_calltf’:
hello.c:8:30: warning: unused parameter ‘user_data’ [-Wunused-parameter]
    8 | static int hello_calltf(char*user_data)
      |                         ~~~~~^~~~~~~~~
hello.c: At top level:
hello.c:14:6: warning: function declaration isn’t a prototype [-Wstrict-prototypes]
   14 | void hello_register()
      |      ^~~~~~~~~~~~~~
hello.c:27:1: warning: function declaration isn’t a prototype [-Wstrict-prototypes]
   27 | void (*vlog_startup_routines[])() = {
      | ^~~~
Making hello.vpi from  hello.o...
/usr/lib/gcc/x86_64-pc-cygwin/12/../../../../x86_64-pc-cygwin/bin/ld: C:\GNU\iverilog\lib/libvpi.a(libvpi.o): in function `vpi_register_cb':
C:\Source\iverilog-build\vpi/../../iverilog/vpi/libvpi.c:31:(.text+0x3e): undefined reference to `__imp__assert'
/usr/lib/gcc/x86_64-pc-cygwin/12/../../../../x86_64-pc-cygwin/bin/ld: C:\GNU\iverilog\lib/libvpi.a(libvpi.o): in function `vpi_remove_cb':
C:\Source\iverilog-build\vpi/../../iverilog/vpi/libvpi.c:36:(.text+0x9e): undefined reference to `__imp__assert'
/usr/lib/gcc/x86_64-pc-cygwin/12/../../../../x86_64-pc-cygwin/bin/ld: C:\GNU\iverilog\lib/libvpi.a(libvpi.o): in function `vpi_register_systf':
C:\Source\iverilog-build\vpi/../../iverilog/vpi/libvpi.c:42:(.text+0xfe): undefined reference to `__imp__assert'
/usr/lib/gcc/x86_64-pc-cygwin/12/../../../../x86_64-pc-cygwin/bin/ld: C:\GNU\iverilog\lib/libvpi.a(libvpi.o): in function `vpi_get_systf_info':
C:\Source\iverilog-build\vpi/../../iverilog/vpi/libvpi.c:47:(.text+0x166): undefined reference to `__imp__assert'
/usr/lib/gcc/x86_64-pc-cygwin/12/../../../../x86_64-pc-cygwin/bin/ld: C:\GNU\iverilog\lib/libvpi.a(libvpi.o): in function `vpi_handle_by_name':
C:\Source\iverilog-build\vpi/../../iverilog/vpi/libvpi.c:55:(.text+0x1c6): undefined reference to `__imp__assert'
/usr/lib/gcc/x86_64-pc-cygwin/12/../../../../x86_64-pc-cygwin/bin/ld: C:\GNU\iverilog\lib/libvpi.a(libvpi.o):C:\Source\iverilog-build\vpi/../../iverilog/vpi/libvpi.c:60: more undefined references to `__imp__assert' follow
collect2: error: ld returned 1 exit status

martinwhitaker, 2025-02-12:
It looks like you are using Cygwin. The only Windows build environment that is regularly tested is MSYS2.

From the error messages, I suspect you need to adjust the compiler and/or linker flags from the defaults used by iverilog-vpi. IIRC, for Cygwin you should be using the shell script version of iverilog-vpi, not the compiled executable that is used with the Mingw compilers.

I don't use Windows (except as necessary for the day job), so can't help you further with this.

caryr, 2025-02-13:
I do not experience any issues compiling the example using the commands given after the example code after fixing an Icarus compile issue specific to cygwin.

I installed the latest cygwin on one of my machines. Then fixed a compile issue (cygwin needs to use -std=gnu++11 instead of -std=c++11 to get the strdup() definition: updated configure.ac). cygwin also does not have docopt installed so vvp_reg.py needed to be updated to match what is done for msys2. There were also a few places where the argument to isspace() and isdigit() needed to be cast to int to remove some compile warnings. I also updated the documentation for the VPI usage to remove the compile warnings in that code. Please see the **Using VPI** section on the primary [Icarus documentation site](https://steveicarus.github.io/iverilog/).

cygwin uses the shell script version of iverilog-vpi to compile the VPI code. I also checked this using MSYS2 which uses the program version of iverilog-vpi. Everything worked as expected for both cygwin and MSYS2.

I also verified the full test suite runs cleanly using the cygwin environment, which includes some VPI tests.

I'm not certain where libvpi.a/libvpi.c is coming from and Icarus provides `$fscanf()` already. I think to help you further we will need the exact commands you are running and possibly how Icarus was built. My guess is libvpi.a was not build in a compatible manner and this code is where the issue is.

caryr, 2025-02-13:
Also I have no idea what you are asking regarding the file tree structure. We need specifics to advise you how to manually resolve your issue with that.

nipscernlab, 2025-02-13:
Thank you for de detailed explanation.
I will try the suggestions.
By now, I just want to continue using $fscanf().
Then, I am happy reading this sentence: "Icarus provides $fscanf() already"
Thai is all I need for the moment.
However, when I try to run my testbench, I get this message from vvp:

Compiling VVP ...
 ... VVP file version 12.0 (devel) (s20150603-1539-g2693dd32b)
C:\Sapho\sapho-p\Teste\Projetos\DTW\TopLevel\top_level_tb.v:49: Error: $fopen() is a system function, it cannot be called as a task.
C:\Sapho\sapho-p\Teste\Projetos\DTW\TopLevel\top_level_tb.v:57: Error: $fscanf() is a system function, it cannot be called as a task.
Compile cleanup...
 ... Linking
 ... Removing symbol tables
 ... Compiletf functions
DTW.vvp: Program not runnable, 2 errors.

IThe same code is running in modelsim.
Any idea?

svenka3, 2025-02-13:
FWIW - Iverilog + VPI works like charm on Windows + WSL (Ubuntu). Not sure if that setup would work for you @nipscernlab - I have been very happy with WSL having moved from Cygwin few years ago. And a note to developers/maintainers of this wonderful software - please consider regressing WSL, if there is anything I can do to be of help in that process, am glad to pitch in.

Regards

svenka3, 2025-02-13:
> IThe same code is running in modelsim.

That error message is clear from Icarus - maybe we can downgrade it to a warning, not sure. But fix is easy and aligns with IEEE LRM.

caryr, 2025-02-13:
When you report an error please also include the line of Verilog that is failing along with the error/warning message. `$fopen()` should be called as a function otherwise you do not have access to the file descriptor/MCD that was opened. It's basically useless without the return value other than possibly creating a zero sized file to indicate something to an external monitor script.

If you are using one of the SystemVerilog generations (-g2005-sv or higher) then this error becomes a warning since SystemVerilog allows functions to be called as tasks. You can also suppress the warning by casting the function to void. which makes it clear the function result is not expected to be used and should be compatible with other simulators.

caryr, 2025-02-13:
I've thought of adding WSL to my development/local testing flow at various times since it is likely faster than either cygwin or MSYS2, but as a developer it's not about speed. It's about testing using a breadth of compile environments and to some extent WSL is basically running Ubuntu on a virtual machine and we already extensively test using the LTS versions of Ubuntu. I used to use cygwin regularly for development, but had not used it for close to a decade until yesterday. I only added MSYS2 a few months ago to resolve some issues specific to MSYS2. Most of my development is done on Ubuntu with different VirtualBox machines to handle different variants of Ubuntu along with a BSD environment.

nipscernlab, 2025-02-13:
This is the piece of code:

reg signed [15:0] data = 16'd0;
integer data_in1;
initial data_in1 = $fopen("sinal_harm_q.txt", "r"); 

always @(posedge clk)
begin
	if (rst_proc)
	begin
		$fscanf(data_in1, "%d", data);
		wrreq <= 1'b1;
	end
end

And the error now is only with $fscanf() (not $fopen):

Compiling VVP ...
 ... VVP file version 12.0 (devel) (s20150603-1539-g2693dd32b)
C:\Sapho\sapho-p\Teste\Projetos\DTW\TopLevel\top_level_tb.v:57: Error: $fscanf() is a system function, it cannot be called as a task.
Compile cleanup...
 ... Linking
 ... Removing symbol tables
 ... Compiletf functions
DTW.vvp: Program not runnable, 1 errors.

svenka3, 2025-02-13:
Try:

```verilog

integer scan_result;
always @(posedge clk)
begin
    if (!rst_proc) begin
        if (!$feof(data_in1)) begin
            scan_result = $fscanf(data_in1, "%d", data);  
            wrreq <= 1'b1;
        end else begin
            wrreq <= 1'b0;
        end
    end else begin
        wrreq <= 1'b0;
    end
end
```

nipscernlab, 2025-02-13:
[sinal_harm_q.txt](https://github.com/user-attachments/files/18788660/sinal_harm_q.txt)

This is the file I am trying to read each line to each positive board of the clk

svenka3, 2025-02-13:
Did you do $fopen to that data_in1 file descriptor handle?

nipscernlab, 2025-02-13:
here: 

initial data_in1 = $fopen("sinal_harm_q.txt", "r");

svenka3, 2025-02-13:
Below code works fine with your data (subset):

```verilog
module m;
integer data_in1;
reg clk;
reg rst_proc;
reg wrreq;
reg [31:0] data;

initial begin
  clk = 0;
  rst_proc = 0;
  data_in1 = $fopen("sinal_harm_q.txt", "r");
    if (data_in1 == 0) begin
      $error("Error opening file");
    end
    else begin
      $info("Reading from file: sinal_harm_q.txt ");
    end

    forever #10 clk = ~clk;
end

initial begin
  #100;
  $finish(2);
end

integer scan_result;
always @(posedge clk)
begin
    if (!rst_proc) begin
        if (!$feof(data_in1)) begin
            scan_result = $fscanf(data_in1, "%d", data);  
            $display ("data read: %0d", data);
            wrreq <= 1'b1;
        end else begin
            wrreq <= 1'b0;
        end
    end else begin
        wrreq <= 1'b0;
    end
end
endmodule

```

Log file:
```log
INFO: t1.sv:18: Reading from file: sinal_harm_q.txt
      Time: 0  Scope: m
data read: 26561
data read: 23974
data read: 17754
data read: 11816
data read: 6563
t1.sv:26: $finish(2) called at 100 (1s)
```

caryr, 2025-02-13:
This is no longer a VPI issue and really seems to be a user coding issue, so can we close this. Also the function as task error can be avoided by using a SystemVerilog generation. See my previous comment.

Our preference when debugging is to have a complete simplified example that demonstrates the failure. This allows us to figure out what is being done incorrectly. For your case it is unlikely there is a bug in Icarus.

nipscernlab, 2025-02-13:
It works for me too!
Still did not understand why mine does not.
But now I can merge your piece in my testbench.

thank you!

caryr, 2025-02-13:
> Still did not understand why mine does not.

A learning opportunity

caryr, 2025-02-13:
For example here is how I would have simplified this:
```
module top;
  integer fd, data;
  initial begin
    fd = $fopen("data.txt", "r");
    forever begin
      void'($fscanf(fd, "%d", data));
      if ($feof(fd)) break;
      $display("read data: %0d", data);
    end
  end
endmodule
```
and when run using:
```
iverilog -g2009 tst.v
vvp a.out
```
produces the following:
```
read data: 26561
read data: 23974
read data: 17754
read data: 11816
:
:
read data: 6380
read data: 11387
read data: 18152
read data: 23611
```
There are no error or warning messages during the compile or runtime and the output values exactly match what is in the input file.

caryr, 2025-02-13:
@svenka3 your version likely has a bug when the last value is read since `$feof()` does not return false until after the `$fscanf()` fails to read a value. This says the last value in the file will likely be processed twice.

nipscernlab, 2025-02-14:
I see.
I am going to use all the checking controls now

Thank you

caryr, 2025-02-14:
That is an excellent idea. We often assume everything will work as expected because it normally does, but it's hard to figure out what failed unless the proper work was put into checking for what can go wrong. These failures may not even produce a failure in the simulation (e.g. reusing the last data point, etc.).

svenka3, 2025-02-14:
> It works for me too! Still did not understand why mine does not. But now I can merge your piece in my testbench.
> 
> thank you!

Maybe some race condition among clk, file-open, feof check?

svenka3, 2025-02-14:
> [@svenka3](https://github.com/svenka3) your version likely has a bug when the last value is read since `$feof()` does not return false until after the `$fscanf()` fails to read a value. This says the last value in the file will likely be processed twice.

Good point - to be honest, I created that code using chatGPT just to get a minimalist, working example to address @nipscernlab  reported issue (task/function usage of $fscanf etc.). 

Cheers

caryr, 2025-02-14:
Sometimes all that is needed is to check/assert that the value is read/generated correctly and then you have an indication that an invalid value was assigned to give you the exact point of failure.

caryr, 2025-02-14:
Good ol' chatGPT, it gives you code that may or may not work and almost always doesn't handle the corner cases correctly. It certainly is a nice tool to get a lot of code written quickly, but don't assume it 100% correct without being reviewed and how as humans do we get to the level where we can review the generated code for errors if we have not written and debugged a large amount of code?

Within the last year I had chatGPT give me an incorrect skin depth (off by a couple orders of magnitude) which is a simple constant and formula look up, so make sure to double check anything it says. I was able to teach it what it was doing wrong and eventually it gave the correct result, but to use it effectively we still need to be smarter enough to know when it's giving us an incorrect answer.

It will get there, I just hope we don't have a major catastrophe because someone blindly assumed chatGPT was correct!

closed 2025-03-07 by caryr